gotthardp/lorawan-server #326: Send HTTP POST with handler

jot4p, 2018-03-06:
Hi, 

I'm trying to send HTTP post with your handler:

https://imgur.com/a/tAVi8

On my server, I get the frame:

127.0.0.1 - - [06/Mar/2018:10:20:23 +0000] "POST /?deveui=0004A30B001BB5EA&payload={data} HTTP/1.1" 200 29874 "-" "-"

why data is not set ?

my received frame on server:

Received | Application | DevAddr | MAC | U/L RSSI | U/L SNR | FCnt | Confirm | Port | Data
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
2018-03-06 10:30:57 | parking_frame_parser | 01DDA4ED | 00800000A00005C9 | -93 | 8 | 29 |   | 4 | 080A00001C00ECFA480993

jot4p, 2018-03-06:
Solved... It's necessary add "data" to Uplink Fields of my handler..

gotthardp, 2018-03-06:
Correct. I love the errors that get solved before I have actually time to look at them. :)

closed 2018-03-06 by gotthardp