gotthardp/lorawan-server #327: "cannot_measure_battery" continually displayed

Howlinmoon, 2018-03-06:
* Distribution type: binary or built from source.
Binary - I assume, supplied via Docker pull

* Hardware type: PC Server, Just PC, MacBook, MacServer, Raspberry Pi1/2/3/x, Odroid, etc.
OS type and version. Also, add the output of the uname -a command if you are running on a sort of a Unix/Linux/MacOS system.

Linux server box,
biff@obsidian:~$ uname -a
Linux obsidian 4.4.0-104-generic #127-Ubuntu SMP Mon Dec 11 12:16:42 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

* Your Erlang/OTP version. Can be obtained by running erl at the command line: please provide all the text before 1> line.
Whatever version is supplied via docker

Background:

I performed the "Docker Pull"  to install it yesterday:  2018/03/05

biff@obsidian:~$ sudo docker pull gotthardp/lorawan-server:latest
[sudo] password for biff: 
latest: Pulling from gotthardp/lorawan-server
4176fe04cefe: Pull complete 
87bf17fdba02: Pull complete 
e0fcc5c11314: Pull complete 
d16fbb5b0ffd: Pull complete 
5977e59576d8: Pull complete 
Digest: sha256:b49c2cffd6c11b4b39519d7e7b69c95b078600b2425f83eb1fb031550e2a6239
Status: Downloaded newer image for gotthardp/lorawan-server:latest

and am running it via:

biff@obsidian:~$ sudo docker run --detach --name lorwan --hostname lorawan --publish 8080:8080/tcp --publish 1680:1680/udp gotthardp/lorawan-server:latest
ead10a0a57a6123f8819553c6868690901526bdc554fdb7eff8f87a9632d4a14

I am using an EU gateway with an EU node.
The node is sending a data stream that cycles like this:
(this is actually sent as 16 byte binary stream)
"\20\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00"
"\00\20\00\00\00\00\00\00\00\00\00\00\00\00\00\00"
"\00\00\20\00\00\00\00\00\00\00\00\00\00\00\00\00"
...
"\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\20"

and then back to the top...

With this setup - the battery level is either 00 or 20 and when I view my node it continually states that the battery level is 255 and displays the tag "cannot_measure_battery"

Am I doing something wrong?
[image]
[image]

gotthardp, 2018-03-06:
Hello. The battery level is determined using a LoRaWAN MAC Command (DevStatusReq) and your device apparently returns 255. This means "The end-device was not able to measure the battery level.", which is exactly what the error message says.

Howlinmoon, 2018-03-07:
Yes, the device I am using does not appear to support the battery level feature, I was under the mistaken impression that was coming from the 16 byte message my device was sending since one field in here is denoted as was denoted as "batteryLevel": https://github.com/Lora-net/LoRaMac-node/blob/master/src/apps/LoRaMac/classA/LoRaMote/main.c#L207

closed 2018-03-07 by Howlinmoon