gotthardp/lorawan-server #329: Users authentication strategy

Quentintin, 2018-03-07:
Hello,

I would like to interact with Network server's users in third party app.

### The main goal
Use a custom strategy of authentication based on Network server's users for my **node-red** environment.

Currently, I succeed to use PAM authentication (based on Linux System users). It works fine.
But i would like to use the same accounts for Network server and Node-Red
- System users
- Custom users
   - Lorawan-server
   - Node-red

Can you provide me any information about users database, any advice to build a mechanism of authentication made for Mnesia ?

Thank you :)

gotthardp, 2018-03-07:
So you want lorawan-server to authenticat some 3rd paty web-sites (node-red in this case)? The password stored is encypted in the form of HTTP Digest's HA1, so this may be a challenge, although not impossible to do.

Quentintin, 2018-03-08:
Can you explain me how to access this stored hashed password ?

Quentintin, 2018-03-08:
I'm a very novice in Erlang (Me + ERL == 0) 
Does Erlang is the only way to access erlang database Mnesia ? 
Which commands can I write to access database directly in command line into the linux shell ?

gotthardp, 2018-03-08:
The database can be accessed from Erlang only. To access it from CMD line you should run the server and the connect to it from another console by `erl -sname test -remsh lorawan@<hostname>`, where `<hostname>` is your hostname.
https://github.com/gotthardp/lorawan-server/blob/master/doc/Development.md#connections-to-the-lorawan-node

Then do `mnesia:dirty_read(users, <<"username">>)`, where username is your username.
http://erlang.org/doc/man/mnesia.html#dirty_read-2

Quentintin, 2018-03-09:
I didn't succes to connect on node with `erl -sname test -remsh lorawan@<myhostname>`

I get the following response : 
```
Erlang/OTP 20 [erts-9.1] [source] [smp:4:4] [ds:4:4:10] [async-threads:10] [kernel-poll:false]

*** ERROR: Shell process terminated! (^G to start new job) ***
```

I try to find any solution about this error on Slack but nothing ...

- Can i have a problem executing command line through SSH connection ?
- my hostname includes hyphen (-) symbol ! Is it problematic ?

gotthardp, 2018-03-09:
You must run this on the same machine where the lorawan-server is already running.
What *exactly* did you type to the SSH, please?

gotthardp, 2018-03-12:
Its also important to know under what user is the server running. You have to make sure that the content of $HOME/.erlang.cookie is the same for the lorawan-server instance as well as for your shell command.

gotthardp, 2018-08-01:
Closing an inactive discussion thread.

closed 2018-08-01 by gotthardp