gotthardp/lorawan-server #332: Segmentation fault

pdl86, 2018-03-10:
Hello 

we have rebuid a new lorawan server based on mlinux 3.3.22 ( latest version) and Lorawan 0.5.3 .

will trying to relaod a dbexport , I've got the following errors on each file : 
/usr/lib/lorawan-server/bin/dbimport: line 11:   778 Segmentation fault      wget -nv $AUTH --header=Content-Type:application/json --post-file=$f http://$SERVER/api/$(basename $f .json)
/usr/lib/lorawan-server/bin/dbimport: line 11:   780 Segmentation fault      wget -nv $AUTH --header=Content-Type:application/json --post-file=$f http://$SERVER/api/$(basename $f .json)
/usr/lib/lorawan-server/bin/dbimport: line 11:   782 Segmentation fault      wget -nv $AUTH --header=Content-Type:application/json --post-file=$f http://$SERVER/api/$(basename $f .json)
/usr/lib/lorawan-server/bin/dbimport: line 11:   784 Segmentation fault      wget -nv $AUTH --header=Content-Type:application/json --post-file=$f http://$SERVER/api/$(basename $f .json)
/usr/lib/lorawan-server/bin/dbimport: line 11:   786 Segmentation fault      wget -nv $AUTH --header=Content-Type:application/json --post-file=$f http://$SERVER/api/$(basename $f .json)
/usr/lib/lorawan-server/bin/dbimport: line 11:   788 Segmentation fault      wget -nv $AUTH --header=Content-Type:application/json --post-file=$f http://$SERVER/api/$(basename $f .json)
/usr/lib/lorawan-server/bin/dbimport: line 11:   790 Segmentation fault      wget -nv $AUTH --header=Content-Type:application/json --post-file=$f http://$SERVER/api/$(basename $f .json)
/usr/lib/lorawan-server/bin/dbimport: line 11:   792 Segmentation fault      wget -nv $AUTH --header=Content-Type:application/json --post-file=$f http://$SERVER/api/$(basename $f .json)

( got the same erro on dbexport through)

any ideas ? 
regards
philippe

gotthardp, 2018-03-10:
Can you try running `wget` as a standalone command?

pdl86, 2018-03-10:
Hi petr

will do monday ( I'm off site)

will let you know

regards

¨Philippe

pdl86, 2018-03-12:
Hello 

same error : 
root@mtcdt:/var/lib/lorawan-server# wget -nv --user=admin --password=admin  --header=Content-Type:application/json --post-file=./backup-CH2i/handlers.json  http://localhost:8080/api/handlers
Segmentation fault

but wget works for a external fetch 

root@mtcdt:/var/lib/lorawan-server# wget google.fr
--2018-03-12 13:12:34--  http://google.fr/
Resolving google.fr... 216.58.198.195, 2a00:1450:4007:809::2003
Connecting to google.fr|216.58.198.195|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://www.google.fr/ [following]
--2018-03-12 13:12:34--  http://www.google.fr/
Resolving www.google.fr... 216.58.198.195, 2a00:1450:4007:809::2003
Reusing existing connection to google.fr:80.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'index.html.1'

    [ <=>                                                                                                                                                                                         ] 11,363      --.-K/s   in 0.004s

2018-03-12 13:12:34 (2.42 MB/s) - 'index.html.1' saved [11363]

regards
Philippe

Apaisal, 2018-03-12:
Hi Phillippe,

Could you use strace with wget command issue? And show your strace log what
is it happen.

Regards,
Anol

gotthardp, 2018-03-12:
This is not a lorawan-server issue. Feels like a faulty Conduit image. Try what @Apaisal is suggesting, please.

pdl86, 2018-03-12:
hi

on my way to do it, but need first to add strace to the image.
will let you know tomorrow.
regardd
philippe

pdl86, 2018-03-13:
Hi 
here is the strace

strace wget -nv --user=admin --password=admin --header=Content-Type:application/json --post-file=./backup-CH2i/handlers.json http://localhost:8080/api/handlers
execve("/usr/bin/wget", ["wget", "-nv", "--user=admin", "--password=admin", "--header=Content-Type:applicatio"..., "--post-file=./backup-CH2i/handle"..., "http://localhost:8080/api/handle"...], [/* 16 vars */]) = 0
brk(0)                                  = 0x7b000
uname({sys="Linux", node="mtcdt", ...}) = 0
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=5247, ...}) = 0
mmap2(NULL, 5247, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f57000
close(3)                                = 0
open("/usr/lib/libssl.so.1.0.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0$\322\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=324596, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f56000
mmap2(NULL, 356084, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6edd000
mprotect(0xb6f27000, 32768, PROT_NONE)  = 0
mmap2(0xb6f2f000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4a000) = 0xb6f2f000
close(3)                                = 0
open("/usr/lib/libcrypto.so.1.0.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\240%\4\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=1449172, ...}) = 0
mmap2(NULL, 1493840, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6d70000
mprotect(0xb6ebc000, 28672, PROT_NONE)  = 0
mmap2(0xb6ec3000, 94208, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x14b000) = 0xb6ec3000
mmap2(0xb6eda000, 11088, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6eda000
close(3)                                = 0
open("/lib/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\344\30\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=78004, ...}) = 0
mmap2(NULL, 109332, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6d55000
mprotect(0xb6d68000, 28672, PROT_NONE)  = 0
mmap2(0xb6d6f000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x12000) = 0xb6d6f000
close(3)                                = 0
open("/usr/lib/libpcre.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\360\f\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=139016, ...}) = 0
mmap2(NULL, 170504, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6d2b000
mprotect(0xb6d4d000, 28672, PROT_NONE)  = 0
mmap2(0xb6d54000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x21000) = 0xb6d54000
close(3)                                = 0
open("/lib/tls/v5l/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/tls/v5l", 0xbe9b6310)      = -1 ENOENT (No such file or directory)
open("/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/tls", 0xbe9b6310)          = -1 ENOENT (No such file or directory)
open("/lib/v5l/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/v5l", 0xbe9b6310)          = -1 ENOENT (No such file or directory)
open("/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0l\204\1\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=1263188, ...}) = 0
mmap2(NULL, 1299872, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6bed000
mprotect(0xb6d1e000, 28672, PROT_NONE)  = 0
mmap2(0xb6d25000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x130000) = 0xb6d25000
mmap2(0xb6d28000, 9632, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6d28000
close(3)                                = 0
open("/lib/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0h\t\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=9752, ...}) = 0
mmap2(NULL, 41144, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6be2000
mprotect(0xb6be4000, 28672, PROT_NONE)  = 0
mmap2(0xb6beb000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0xb6beb000
close(3)                                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f55000
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f54000
set_tls(0xb6f544c0, 0xb6f54ba8, 0xb6f5b058, 0xb6f544c0, 0xb6f5b058) = 0
mprotect(0xb6d25000, 8192, PROT_READ)   = 0
mprotect(0xb6beb000, 4096, PROT_READ)   = 0
mprotect(0xb6f5a000, 4096, PROT_READ)   = 0
munmap(0xb6f57000, 5247)                = 0
[... 76 lines cut ...]
fstat64(3, {st_mode=S_IFREG|0755, st_size=42656, ...}) = 0
mmap2(NULL, 74672, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6bcf000
mprotect(0xb6bd9000, 28672, PROT_NONE)  = 0
mmap2(0xb6be0000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0xb6be0000
close(3)                                = 0
mprotect(0xb6be0000, 4096, PROT_READ)   = 0
munmap(0xb6f57000, 5247)                = 0
open("/etc/hosts", O_RDONLY|O_CLOEXEC)  = 3
fcntl64(3, F_GETFD)                     = 0x1 (flags FD_CLOEXEC)
[... cut ...]
connect(3, {sa_family=AF_INET, sin_port=htons(8080), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
select(4, NULL, [3], NULL, {900, 0})    = 1 (out [3], left {899, 999965})
write(3, "POST /api/handlers HTTP/1.1\r\nUse"..., 180) = 180
select(4, [3], NULL, NULL, {900, 0})    = 1 (in [3], left {899, 999966})
recv(3, "HTTP/1.1 401 Unauthorized\r\nconte"..., 511, MSG_PEEK) = 192
read(3, "HTTP/1.1 401 Unauthorized\r\nconte"..., 192) = 192
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0} ---
+++ killed by SIGSEGV +++
Segmentation fault

pdl86, 2018-03-13:
Hi 

may this can have a link 

few seconds after reboot, I've got ( once) this message on teh console : 

jffs2: warning: (773) jffs2_sum_write_data: Not enough space for summary, padsize = -42

any clue  ?
regards
Philippe

Apaisal, 2018-03-13:
Hi Philippe,

May this line is your issue.

write(2, "POST data file './backup-CH2i/ha"..., 80POST data file './backup-CH2i/handlers.json' missing: No such file or directory) = 80

Could you check file exist or use absolute file path?

gotthardp, 2018-04-05:
Closing due to inactivty. This was not lorawan-server related issue anyway.

pdl86, 2018-04-06:
Sorry I was abroad for a long time.

note taht the issue arises on dbexport AND dbimport.

so as for export , this is not an issue with a non existent file

on the below trace for an export, looks like there are missing libraries ? 
strace  wget -nv --user=admin --password=admin --output-document=backup-20180406-125148/connectors.json http://localhost:8080/api/connectors
execve("/usr/bin/wget", ["wget", "-nv", "--user=admin", "--password=admin", "--output-document=backup-2018040"..., "http://localhost:8080/api/connec"...], [/* 16 vars */]) = 0
brk(0)                                  = 0x7b000
uname({sys="Linux", node="mtcdt", ...}) = 0
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=5247, ...}) = 0
mmap2(NULL, 5247, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f23000
close(3)                                = 0
open("/usr/lib/libssl.so.1.0.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0$\322\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=324596, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f22000
mmap2(NULL, 356084, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ea9000
mprotect(0xb6ef3000, 32768, PROT_NONE)  = 0
mmap2(0xb6efb000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4a000) = 0xb6efb000
close(3)                                = 0
open("/usr/lib/libcrypto.so.1.0.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\240%\4\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=1449172, ...}) = 0
mmap2(NULL, 1493840, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6d3c000
mprotect(0xb6e88000, 28672, PROT_NONE)  = 0
mmap2(0xb6e8f000, 94208, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x14b000) = 0xb6e8f000
mmap2(0xb6ea6000, 11088, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ea6000
close(3)                                = 0
open("/lib/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\344\30\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=78004, ...}) = 0
mmap2(NULL, 109332, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6d21000
mprotect(0xb6d34000, 28672, PROT_NONE)  = 0
mmap2(0xb6d3b000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x12000) = 0xb6d3b000
close(3)                                = 0
open("/usr/lib/libpcre.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\360\f\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=139016, ...}) = 0
mmap2(NULL, 170504, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6cf7000
mprotect(0xb6d19000, 28672, PROT_NONE)  = 0
mmap2(0xb6d20000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x21000) = 0xb6d20000
close(3)                                = 0
open("/lib/tls/v5l/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/tls/v5l", 0xbece1340)      = -1 ENOENT (No such file or directory)
open("/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/tls", 0xbece1340)          = -1 ENOENT (No such file or directory)
open("/lib/v5l/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/v5l", 0xbece1340)          = -1 ENOENT (No such file or directory)
open("/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0l\204\1\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=1263188, ...}) = 0
mmap2(NULL, 1299872, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6bb9000
mprotect(0xb6cea000, 28672, PROT_NONE)  = 0
mmap2(0xb6cf1000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x130000) = 0xb6cf1000
mmap2(0xb6cf4000, 9632, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6cf4000
close(3)                                = 0
open("/lib/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0h\t\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=9752, ...}) = 0
mmap2(NULL, 41144, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6bae000
mprotect(0xb6bb0000, 28672, PROT_NONE)  = 0
mmap2(0xb6bb7000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0xb6bb7000
close(3)                                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f21000
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f20000
set_tls(0xb6f204c0, 0xb6f20ba8, 0xb6f27058, 0xb6f204c0, 0xb6f27058) = 0
[... 83 lines cut ...]
munmap(0xb6f23000, 5247)                = 0
open("/etc/hosts", O_RDONLY|O_CLOEXEC)  = 4
fcntl64(4, F_GETFD)                     = 0x1 (flags FD_CLOEXEC)
fstat64(4, {st_mode=S_IFREG|0644, st_size=72, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f24000
read(4, "# IP-Address Full-Qualified-Host"..., 4096) = 72
read(4, "", 4096)                       = 0
close(4)                                = 0
munmap(0xb6f24000, 4096)                = 0
socket(PF_INET, SOCK_STREAM, IPPROTO_IP) = 4
[... cut ...]
select(5, NULL, [4], NULL, {900, 0})    = 1 (out [4], left {899, 999967})
write(4, "GET /api/connectors HTTP/1.1\r\nUs"..., 130) = 130
select(5, [4], NULL, NULL, {900, 0})    = 1 (in [4], left {899, 992971})
recv(4, "HTTP/1.1 401 Unauthorized\r\nconte"..., 511, MSG_PEEK) = 192
read(4, "HTTP/1.1 401 Unauthorized\r\nconte"..., 192) = 192
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0} ---
+++ killed by SIGSEGV +++
Segmentation fault

so it is more related to the build of the  mtcdt OS ? 

any help apreciated 
regards
Philippe

gotthardp, 2018-04-06:
Interestingly enough, I got a SIGSEGV too on one machine, but not on another machine. Will try to investigate a bit more.

gotthardp, 2018-05-18:
At least in my case the SIGSEGV is due to this bug https://bugzilla.redhat.com/show_bug.cgi?id=1099680,
which has been fixed. I recommend updating your OS or at least the wget.
I am closing this issue as non-server related.

closed 2018-05-18 by gotthardp

pdl86, 2018-05-18:
Thanks for the info

unfortunately , as a first approach , updating  wget , on mlinux  ( via
opkg ) does not give anything as claiming to be up-to-date

so I need to find a way to update the wget sources in the mlinux
firmware building process.

regards
philippe