gotthardp/lorawan-server #333: Callback server compatibity

jot4p, 2018-03-12:
Hi,

I opened a new topic because I think many others have this doubt:

What are the steps to get a POST and get data available for a downlink?

For example:
At the windows command prompt:

```
curl -H "Content-Type: application/json" -X POST http://localhost:8080/callback/AAAAAAA -d "{\"data\":\"ABCDEFG\"}"
``` 

will this command send a post to a handler?

it's necessary create a connection with the "received topic" : **/callback/{devaddr}**^

And after that ?  I'm wrong ?

Please I need documentation to do this..

gotthardp, 2018-03-14:
Your example does not have Authentication. The following works for me:

```
curl -v --digest -uadmin:admin -H "Content-Type: application/json" -X POST http://localhost:8080/callback/11223344 -d "{\"data\":\"ABCDEFG\"}"
```

jot4p, 2018-03-14:
Yes.. I solve the problem. The problem is that user and pw is based on user of server not user of handler.

gotthardp, 2018-03-14:
I just updated the documentation to make it a bit clearer to other.

closed 2018-03-14 by gotthardp