gotthardp/lorawan-server #334: How add network id in uplink field?

Apaisal, 2018-03-13:
Hi @gotthardp 

Could you advise how add network id in uplink field?

gotthardp, 2018-03-13:
Hello. This is not possible unfortunatelly.

gotthardp, 2018-03-15:
Just a question @Apaisal , by "network id" you mean the name of the Network or the NetID parameter? What would you need to add and why?

Apaisal, 2018-03-15:
Yes, I mean netid. It will represent name of the network.
I just interested multiple gateways in one network. Gateway has limitation
on area but network won't.
You can imagine it as Network Domain with GW group.

For example  ->  A network has 2 GW. First gateway in USA and second GW in
Europe. I called "LoRa Network Domain".

gotthardp, 2018-03-15:
You know that for the OTAA devices that jpined over my server the upper bits of DevAddr correspond to the netit. That is not enough? You really need the netid?

Apaisal, 2018-03-16:
I think, OTAA may be possible. But ABP was odd. Deveui insufficient for
huge massive who implement ABP. ABP is locked device to the network and ABP
device to be moved between networks.

gotthardp, 2018-03-30:
I just added netid to the uplink fields. Will be in 0.5.4.

closed 2018-03-30 by gotthardp