gotthardp/lorawan-server #336: "downlink_missed" problem

jot4p, 2018-03-14:
If you check my log:

```
2018-03-13 11:36:59.465 [debug] <0.25220.0>@lorawan_mac_commands:build_fopts:73 018F0C23 <- [dev_status_req]
2018-03-13 11:37:03.612 [debug] <0.25226.0>@lorawan_mac_commands:handle_fopts:16 018F0C23 -> [{dev_status_ans,255,4}]
2018-03-13 11:37:03.612 [debug] <0.25226.0>@lorawan_mac_commands:handle_status:218 DevStatus: battery 255, margin: 4 (max -12.5)
2018-03-13 11:37:08.524 [warning] <0.25230.0> node 018F0C23 downlink_missed
2018-03-13 11:37:08.721 [debug] <0.25230.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:13.488 [warning] <0.25232.0> node 018F0C23 downlink_missed
2018-03-13 11:37:13.686 [debug] <0.25232.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:18.464 [warning] <0.25234.0> node 018F0C23 downlink_missed
2018-03-13 11:37:18.670 [debug] <0.25234.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:23.462 [warning] <0.25238.0> node 018F0C23 downlink_missed
2018-03-13 11:37:23.669 [debug] <0.25238.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:27.679 [warning] <0.25240.0> node 018F0C23 downlink_missed
2018-03-13 11:37:27.886 [debug] <0.25240.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:32.836 [warning] <0.25242.0> node 018F0C23 downlink_missed
2018-03-13 11:37:33.052 [debug] <0.25242.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:38.170 [warning] <0.25245.0> node 018F0C23 downlink_missed
2018-03-13 11:37:38.376 [debug] <0.25245.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:42.568 [warning] <0.25247.0> node 018F0C23 downlink_missed
2018-03-13 11:37:42.768 [debug] <0.25247.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:48.355 [warning] <0.25249.0> node 018F0C23 downlink_missed
2018-03-13 11:37:48.557 [debug] <0.25249.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:37:54.525 [warning] <0.25251.0> node 018F0C23 downlink_missed
2018-03-13 11:37:54.731 [debug] <0.25251.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:38:00.683 [warning] <0.25254.0> node 018F0C23 downlink_missed
2018-03-13 11:38:00.893 [debug] <0.25254.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:38:06.846 [warning] <0.25256.0> node 018F0C23 downlink_missed
2018-03-13 11:38:07.049 [debug] <0.25256.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
2018-03-13 11:38:13.013 [warning] <0.25258.0> node 018F0C23 downlink_missed
2018-03-13 11:38:13.210 [debug] <0.25258.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
 ```

It's possible to check that sensor return the response:

```
2018-03-13 11:37:03.612 [debug] <0.25226.0>@lorawan_mac_commands:handle_fopts:16 018F0C23 -> [{dev_status_ans,255,4}]
2018-03-13 11:37:03.612 [debug] <0.25226.0>@lorawan_mac_commands:handle_status:218 DevStatus: battery 255, margin: 4 (max -12.5)
```
but after that retrurns:
``` 
2018-03-13 11:37:18.464 [warning] <0.25234.0> node 018F0C23 downlink_missed
2018-03-13 11:37:18.670 [debug] <0.25234.0>@lorawan_handler:retransmit:210 018F0C23 retransmitting
```

My json frame is based on  this:

```
{
  ["deveui"]=>
  string(16) "0004A30B001C199E"
  ["data"]=>
  string(16) "12030D020B070000"
  ["confirmed"]=>
  bool(true)
}
```

some trials:
 this:
 ```
 fun(Fields, Data) ->
   Fields#{data => lorawan_utils:binary_to_hex(Data)}
 end.
 ```
return:
 ```
 2018-03-13 16:18:07.774 [error] <0.32391.0> handler 7061726B696E675F6672616D655F706172736572 {build_failed,{error,{badarity,{#Fun<erl_eval.12.99386804>,[#{confirmed => true,data => <<18,3,13,2,17,18,0,0>>,devaddr => <<1,143,12,35>>}]}}}}
 ```

I need to get this frame... and send "data" for devaddr or deveui 0004A30B001C199E. 

How to do this ?

gotthardp, 2018-03-14:
What does your device log say? Did the device receive the downlink? Did the device set the ACK flag?

jot4p, 2018-03-14:
It's not possible to get device log. 

I change the uplink to:
```
fun(_Port, <<Count, Data:Count/binary>>) ->
#{data => binary_to_list(Data)}
end.
```
and downlink to:
```
fun(Data) ->
  #{data=> Data}
end.
```
And my log is:

```
2018-03-14 17:28:17.100 [info] <0.2946.0> device 0004A30B001C199E {join,<<"018F0C23">>}
2018-03-14 17:28:17.100 [debug] <0.2946.0>@lorawan_mac:encode_accept:392 Join-Accept <<"018F0C23">>, netid <<0,0,0>>, cflist [], rx1droff 0, rx2dr 4, appkey <<"00800000A00005C900800000A00005C9">>, appnce <<"811CB3">>
2018-03-14 17:28:17.100 [debug] <0.2946.0>@lorawan_handler:join:107 Join-Accept in RX1: {rxq,868.1,<<"SF7BW125">>,<<"4/5">>,{{2018,3,14},{17,32,28.959291}},3425693323,-94,9.2}
2018-03-14 17:28:17.534 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_down,<0.2698.0>,http,normal,[],[]}
2018-03-14 17:28:18.503 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_down,<0.2671.0>,http,normal,[],[]}
2018-03-14 17:28:19.503 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_down,<0.2668.0>,http,normal,[],[]}
2018-03-14 17:28:20.509 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_down,<0.2665.0>,http,normal,[],[]}
2018-03-14 17:28:21.753 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_down,<0.2662.0>,http,normal,[],[]}
2018-03-14 17:28:22.031 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_up,<0.2657.0>,http}
2018-03-14 17:28:22.548 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_up,<0.2698.0>,http}
2018-03-14 17:28:23.517 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_up,<0.2671.0>,http}
2018-03-14 17:28:24.516 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_up,<0.2668.0>,http}
2018-03-14 17:28:25.013 [debug] <0.2947.0>@lorawan_mac_commands:store_actual_adr:138 DataRate 018F0C23 switched to dr 3
2018-03-14 17:28:25.013 [debug] <0.2947.0>@lorawan_mac_commands:build_fopts:73 018F0C23 <- [dev_status_req]
2018-03-14 17:28:25.517 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_up,<0.2665.0>,http}
2018-03-14 17:28:26.779 [debug] <0.312.0>@lorawan_connector_http:handle_info:137 Unknown message: {gun_up,<0.2662.0>,http}
2018-03-14 17:28:29.149 [debug] <0.2952.0>@lorawan_mac_commands:handle_fopts:16 018F0C23 -> [{dev_status_ans,255,7}]
2018-03-14 17:28:29.149 [debug] <0.2952.0>@lorawan_mac_commands:handle_status:218 DevStatus: battery 255, margin: 7 (max -12.5)
2018-03-14 17:28:29.149 [error] <0.2952.0>@lorawan_mac:cipher:492 gen_statem <0.2952.0> in state uplink terminated with reason: no function clause matching lorawan_mac:cipher(#{data => #{data => <<18,3,14,3,18,28,0,0>>,devaddr => <<1,143,12,35>>}}, <<0,190,121,129,121,165,200,20,121,55,249,183,129,114,10,254>>, 1, <<1,143,12,35>>, 2, 1, <<>>) line 492
2018-03-14 17:28:29.149 [error] <0.2952.0>@lorawan_mac:cipher:492 CRASH REPORT Process <0.2952.0> with 0 neighbours crashed with reason: no function clause matching lorawan_mac:cipher(#{data => #{data => <<18,3,14,3,18,28,0,0>>,devaddr => <<1,143,12,35>>}}, <<0,190,121,129,121,165,200,20,121,55,249,183,129,114,10,254>>, 1, <<1,143,12,35>>, 2, 1, <<>>) line 492
2018-03-14 17:28:29.149 [error] <0.304.0>@lorawan_mac:cipher:492 Supervisor lorawan_handler_sup had child handler started with lorawan_handler:start_link() at <0.2952.0> exit with reason no function clause matching lorawan_mac:cipher(#{data => #{data => <<18,3,14,3,18,28,0,0>>,devaddr => <<1,143,12,35>>}}, <<0,190,121,129,121,165,200,20,121,55,249,183,129,114,10,254>>, 1, <<1,143,12,35>>, 2, 1, <<>>) line 492 in context child_terminated

```

gotthardp, 2018-03-14:
Could you remove both uplink and downlink functions completely? Just leave the field blank. They are optional and for now you don't need them.

jot4p, 2018-03-14:
Ok.... after reboot server... maybe the problema is "visual" 

Server log: https://imgur.com/hWPGj5K 

Debug.log: https://imgur.com/QyYcBbh 

Visual : https://imgur.com/a/WAiow

gotthardp, 2018-03-14:
Everything looks good. What is not happening that you expect to happen?

jot4p, 2018-03-15:
Ok.. the problem is based on my "MQTT publish" in your server.. 

I'm using the same handler called "parking_sensor_parser" to send the frames from nodes to callback server  and for MQTT..   but the problem is here. 

If I remove MQTT from this handler.. I can receive the "CONFIRM FRAME" but if I use at same time i can't receive the frame. 

https://imgur.com/a/Fd5Ox  << error with this , why ?

jot4p, 2018-03-15:
Fixed. 

The problem is that I have subscribed the same Topic that I publish and created :)

Thanks.

closed 2018-03-15 by jot4p