gotthardp/lorawan-server #337: Uplinks Fields are not saved on the Handlers screen

glaucogomes, 2018-03-15:
Hi @gotthardp,

I upgraded to version 5.0.4 and had two issues with the Handlers:

1 - problem to create and edit handlers: Uplink Fields are not saved;

2 - the Payload combobox in Handler screen don't show the ASCII Text option.

My connectors also stopped working, but I believe it was because of this problem with the Handlers.

Thanks.

gotthardp, 2018-03-15:
Hello. I cannot reproduce this. Also, other users work with this and didn't report any error.

glaucogomes, 2018-03-15:
I restored the database backup and everything worked perfectly. Magic things, my bad. I'll try harder next time before opening an issue. Thank you!

gotthardp, 2018-03-15:
No problem. I'm glad it works.

closed 2018-03-15 by gotthardp