gotthardp/lorawan-server #338: Integration with cayenne

qrimly, 2018-03-16:
Hi Gotthardp,

i have read rules how to connect lora network server to cayenne application server through MQTT, but the problem is to upload one data sensor, we must set new topic or slash (https://mydevices.com/cayenne/docs/cayenne-mqtt-api/#cayenne-mqtt-api-manually-publishing-subscribing-send-sensor-data-to-cayenne) , but in network server that you made, we cannot (that i know) set many topic for one handlers. so can you tell me to solve this problem?

Thanks.

gotthardp, 2018-03-16:
But you can use a template, i.e. set the topic to `some/{field}` where `field` is one of the fields; this way you can send data to various topics even from one handler/connector. would this be sufficient or you need something extra?

qrimly, 2018-03-16:
thank you for the answer. can you give me example please? for example i have data payload 3bytes (00FF50) 1 and 2 for temperature at chanel 0, and 3 for humidity at channel 1, how to i write the function on parse uplink?

gotthardp, 2018-03-16:
So the 3 bytes is a temperature + humidity and you want each to send to a different topic?

qrimly, 2018-03-16:
yes, like format on cayenne, each topic mus sen data format like this

(string) type,unit=value

example for temperature = temp,c=27.3
example for humidity       = rel_hum,p=50

gotthardp, 2018-03-16:
But this is not the cayenne binary format, right? It does not include the channel IDs.

qrimly, 2018-03-19:
that the cayenne publish format, you can see how to publishing or subscribing using manual MQTT (https://mydevices.com/cayenne/docs/cayenne-mqtt-api/#cayenne-mqtt-api-manually-publishing-subscribing-send-sensor-data-to-cayenne) the formta to send data must like this, and i don't know how to convert from backend data to cayenne format, please guide me to convert the data.

Zupayruna1, 2022-08-17:
I had problem with cayenne default decoder inside compact loraserver (maybe is not the current version), if ur working in nodered u can use a cayenne node decoder and u can see ur data