gotthardp/lorawan-server #339: Feature Request: Connector with multiple Applications

webfrank, 2018-03-16:
I think it would be more convenient to have the possibility to link multiple applications to a single connector instead of creating one connector per application which in turn is linked with one profile.
If relations should be 1:1 all the configuration could be done on a single place.

altishchenko, 2018-04-04:
Copy on that. It was standard behaviour for the 0.4.13, where you could have multiple different devices with there custom handlers(applications) talking with the external broker through single connector...

xypron, 2019-04-27:
I have different things which need different value mapping. So I need multiple handlers. [io.adafruit.com](https://io.adafruit.com) accepts only a single concurrent connection. 

So I really need the single connector to be mapped to multiple handlers.