gotthardp/lorawan-server #340: Priority between acknowledgment and downlink

cduplouy, 2018-03-21:
Hi,

I have some troubles in the experimentation I am running and I was wondering if any of you had a solution to that problem.

To be concise, I have my class C devices that are continually waiting to receive data on a multicast channel. I use a python script to communicate via a ws with the gateway to send a downlink with : 
{"devaddr":"ABCD0123","data":"01", "port":2,"time":"immediately"}
When the device get the data 0x01, it makes a led blink.

But every 10 minutes, the devices send a keepalive and expect to receive an acknowledgment. Thus, that's the only moment when the devices are not listening for data on the multicast channel.

The problem is that if I try to send this downlink when a device already sent the keepalive and is waiting for the acknowledgment, it wont get the data to make the led blink.

I tried to send 4 downlinks 4-5 seconds apart to be sure that the device would get the data at least once, but the gateway first sends all the downlinks before sending the acknowledgment to the device.

Is there any way to send the acknowledgment first ?

Thank you !

gotthardp, 2018-03-21:
Interesting problem. The server processing of ACK and class C are independent, which may be the root cause. When you send class C it is immediately sent to the gateway without any checks, so if the device sends something the queue may get confused.

cduplouy, 2018-03-23:
To get over the problem, I implemented a different approach.
My devices send a message without asking/expecting an ACK but I use the multicast channel to notify them that the GW is still alive.

Thanks :)

gotthardp, 2018-03-28:
Cool :)

closed 2018-03-28 by gotthardp