gotthardp/lorawan-server #342: repeated null frame

jot4p, 2018-03-22:
Hi, 

[image]

Sorry, I need to understand what is this ?  The server show this some times ... what is this ?

matiasz8, 2018-03-22:
Hello,
I'd had the same problem. But, before this, I created the Active node, I have the profile and the Application related. 
This start to work when I changed the Frame Counter from "Strict 16bits" to "Disabled".

gotthardp, 2018-03-28:
What is the issue, please? You are seeing repeated transmission of Frame 0. If your Frame Counter is "Strict" and you sent some frames before, this will generate an error. A solution for this is to either Disable the Fcnt check (as suggested by @matiasz8, or to use "reset on zero"

gotthardp, 2018-04-05:
Closing due to inactivity. I assume the issue was clarified / disappeared.

closed 2018-04-05 by gotthardp