gotthardp/lorawan-server #344: Possibly ineffective network ADR parameters

arskanov, 2018-03-23:
Running version 0.5.3 of the debian package on 4.13.0-37-generic #42~16.04.1-Ubuntu SMP  x86_64 GNU/Linux

Lorawan-server is submitting incorrect (out of range) link_adr_req:s to my RN2483 (EU868) modules. This results in a "power" error.

RN2483 supported TXPower modes: 1-5
link_adr_req may request also 0, 6 and 7 Power modes. 

```
2018-03-23 13:48:34.780 [debug] <0.10493.0>@lorawan_mac_commands:send_adr:332 LinkADRReq {0,5,[{0,2}]}
2018-03-23 13:48:34.780 [debug] <0.10493.0>@lorawan_mac_commands:build_fopts:74 C0FFEEEE <- [{link_adr_req,5,0,7,0,0}]
2018-03-23 13:48:39.791 [debug] <0.10517.0>@lorawan_mac_commands:handle_fopts:16 C0FFEEEE -> [{link_adr_ans,0,1,1}]
2018-03-23 13:48:39.792 [warning] <0.10517.0> node C0FFEEEE {adr_req_failed,{0,1,1}}

```

I'm having trouble mapping dBm values in lorawan-server to the corresponding TXPower indexes. If I select the following settings:

Network:
- Maximum EIRP: 16dBm
- Max Power: Max-2dB
- Min Power: Max -8dB

Profile:
- SetPower: 14dBm

Node: 
- Set Power: Manually select 14dBm

Lorawan-server will send out a valid "1" power idx and everything is OK. However I can still select the 16dBm, and cause the "0" power idx to be sent to the node. I'm guessing that this can also be selected by the ADR algorithm.

The same thing can happen on the other end of the spectrum, with lorawan-server determining the best power index to send the node to be 6 or 7, and thus generating an adr NAK.

Setting the network's ADR parameters does somehow affect the Node's dBm <-> Power idx mapping, but the top selectable value always results in a "0" pwridx being sent to the node. This means even if Network: Max EIRP is 14dB, the issue is still present.

I can provide more information if needed, and I do still hope it's just my misunderstanding of the documentation.

gotthardp, 2018-03-28:
Hello. The ADR algorithm will (shall) obey the Max Power settings.

You can still set the Max power manually via the web admin-- the values displayed are not checked against the Max Power (to simplify the implementation). It is the responsibility of the user to remember what was set as Max Power.

arskanov, 2018-03-29:
Ah, I see, so no real problem there.

Thank you for the clarification. I appreciate your work on this project! Works great.

closed 2018-03-29 by arskanov