gotthardp/lorawan-server #346: Unable to modify Gateway settings in Server Admin

romansoft, 2018-03-25:
After making changes to settings fields and pressing Save Changes, an error message appears :Data has changed. Please Reload and repeat the action.". Reloading and repeating causes the same error.

[image]

gotthardp, 2018-03-28:
What server version are you using?

gotthardp, 2018-03-29:
Could your gateway be sending stats more often? The server stores stats, so if you edit a gateway info you should do that before it sends new stats. This is a known limitation.

romansoft, 2018-03-29:
I'm using the gateway PF config settings recommended in the larawan-server documentation. Also, this happens even with gateway being disconnected.

gotthardp, 2018-03-30:
Yeah, you are right. This is related to the "status" field, which includes a disconnection time, increasing every 20 secs. I made a minor enhancement that could reduce the occurrences to a satisfactory level.

closed 2018-03-30 by gotthardp