gotthardp/lorawan-server #348: lorawan-server arm cross compiling

ennasiry, 2018-03-26:
Is it possible to cross compile lorawan-server project using an arm-angstrom-linux-gnueabi env passing by the makefile ? 
thank you

ennasiry, 2018-03-28:
> No cross-compilation is necessary. The server binaries run on every
> platform that has Erlang installed. You can build/compile the binaries on
> any machine and then simply copy them to another machine. As long as the
> target machine uses the same (or higher) Erlang version, everything will
> work.

my host is x86_64 linux computer while my target is an 32 arm linux
embedded board ,thats will still work ?

gotthardp, 2018-03-28:
Yes, that will still work. (Assuming you have Erlang installed on the 32-bit ARM too.)

ennasiry, 2018-03-28:
i have added meta-erlang layer to my embedded linux image and i dont know
how to verify if it is compatible with my host version (erlang /OTP 20)

git clone https://github.com/joaohf/meta-erlang.git

im still confused wich binaries do i have to copy ,and wich one that i
have to excute.

do i have to follow the same procedure mentioned in the installation
section ? ( bin/lorawan-server )

gotthardp, 2018-03-28:
That should work. BTW, are you aware of my meta-lorawan bitbake recipe?
https://github.com/gotthardp/meta-lorawan/blob/master/recipes-core/lorawan-server/lorawan-server_0.5.3.bb

That does everything automatically.

ennasiry, 2018-03-28:
> recipes-core/lorawan-server/lorawan-server_0.5.3.bb

yes i have parsed that recipe and i write my own image recipe for the board
im using but i get an escript error.
( $ which escript
/usr/bin/escript )
/usr/bin/env: ‘escript’: No such file or directory
any idea how to solve that ?

gotthardp, 2018-03-28:
Don't you have Ubuntu by change? :) There is Issue #349 related to escript on Ubuntu.

ennasiry, 2018-03-29:
how can i know if erlang executable files are 32 or 64 bits ? if i copied
the binaries ( that are build and compiled on an 64bits environment ) as
they are to my 32bit arm ( my board is provided with meta erlang layer )
bit and run it ,it will work ?

gotthardp, 2018-03-29:
Yes, that will work. The .beam files (erlang binaries) are the same for 32 and 64 machines.
Did you solve the escript issue?

ennasiry, 2018-03-29:
not really ,i will just copy the project as it is and try to run the
binaries without adding lorawan-server meta to my linux image.
i will let you know if i solved that issue.
thank you

ennasiry, 2018-03-29:
i tried to run it and its giving this error :
{"init terminating in do_boot",{{badmatch,{error,{os_mon,{"no such file or
directory","os_mon.app"}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawa}
init terminating in do_boot
({{badmatch,{error,{os_mon,{_}}}},[{lorawan_app,start,0,[{_},{_}]},{init,start_em,1,[]},{init,do_boot,3,[]}]})

Crash dump is being written to: erl_crash.dump...done

any idea whats the problem ??

2018-03-29 14:49 GMT+01:00 NOUREDDINE ENNASIRY <
ennasiry.noureddine@gmail.com>:

> not really ,i will just copy the project as it is and try to run the
> binaries without adding lorawan-server meta to my linux image.
> i will let you know if i solved that issue.
> thank you
>
>
>
>> Yes, that will work. The .beam files (erlang binaries) are the same for
>> 32 and 64 machines.
>> Did you solve the escript issue?
>>
>>
>
>
>
>
> Cordialement
>
> EN-NASIRY  NOUREDDINE
>
>
> +212 6 27 01 56 91
>
> ennasiry.noureddine@gmail.com

ennasiry, 2018-03-29:
is it necessary to have os_mon to run the project ? i cant find it under
/usr/lib/erlang/lib !

2018-03-29 15:28 GMT+01:00 NOUREDDINE ENNASIRY <
ennasiry.noureddine@gmail.com>:

> i tried to run it and its giving this error :
> {"init terminating in do_boot",{{badmatch,{error,{os_mon,{"no such file
> or directory","os_mon.app"}}}},[{lorawan_app,start,0,[{file,"/
> home/travis/build/gotthardp/lorawa}
> init terminating in do_boot ({{badmatch,{error,{os_mon,{_}
> }}},[{lorawan_app,start,0,[{_},{_}]},{init,start_em,1,[]},{
> init,do_boot,3,[]}]})
>
> Crash dump is being written to: erl_crash.dump...done
>
> any idea whats the problem ??
>
> 2018-03-29 14:49 GMT+01:00 NOUREDDINE ENNASIRY <
> ennasiry.noureddine@gmail.com>:
>
>> not really ,i will just copy the project as it is and try to run the
>> binaries without adding lorawan-server meta to my linux image.
>> i will let you know if i solved that issue.
>> thank you
>>
>>
>>
>>> Yes, that will work. The .beam files (erlang binaries) are the same for
>>> 32 and 64 machines.
>>> Did you solve the escript issue?
>>>
>>>
>>
>>
>>
>>
>> Cordialement
>>
>> EN-NASIRY  NOUREDDINE
>>
>>
>> +212 6 27 01 56 91
>>
>> ennasiry.noureddine@gmail.com
>>
>>
>>
>
>
>
> Cordialement
>
> EN-NASIRY  NOUREDDINE
>
>
> +212 6 27 01 56 91
>
> ennasiry.noureddine@gmail.com

gotthardp, 2018-03-29:
Yes, os_mon is required. It's another erlang-package to install.

ennasiry, 2018-03-29:
that means that meta-erlang is not provided with all necessary packages !
how does it still work for mlinux as it is mentioned at the project
repository ?

gotthardp, 2018-03-29:
but erlang-os-mon is included in meta-erlang and declared as a dependency of the meta-lorawan

ennasiry, 2018-03-30:
solved ,thank you ,its working now !

gotthardp, 2018-03-30:
Cool. What was the solution?

ennasiry, 2018-03-30:
i have wrote my own lorawan-server image recipe ,and i installed erlang
packages on my board using opkg.

closed 2018-03-30 by ennasiry

gotthardp, 2018-03-30:
How is your  lorawan-server image recipe different to mine? Perhaps your's is better :)

rkdm91, 2018-06-20:
Just for people who wants to install the lorawan server from the Binary files on a openembbeded image. All packages needed are include in the meta-erlang but there are not installed by default.
There is two solutions : 
The simplest way  : Add "erlang-modules" instead of "erlang" to your local.conf ( this will add all the modules, PS : not only os_mon is needed )
Or, Add the desired modules by modifying generate-manifest file (meta-erlang/scripts/contrib/erlang/)  and regenerate the manifest.

gotthardp, 2018-06-20:
You say that the erlang dependencies are not installed regardless the RDEPENDS? https://github.com/gotthardp/meta-lorawan/blob/master/recipes-core/lorawan-server/lorawan-server_0.5.5.bb#L17

rkdm91, 2018-06-20:
I was refering to use the Binary release on linux ( https://github.com/gotthardp/lorawan-server/blob/master/doc/Installation.md#using-the-binary-release-on-linux ) with openembedded. In that case Erlang need to be added to the openembeded image, but as ennasiry said if you just installed erlang from meta-erlang (https://github.com/joaohf/meta-erlang.git) all the modules needed are not installed ( as os_mon) that why he couldn't find os_mon on /usr/lib/erlang/lib.

ennasiry, 2018-06-20:
yes those modules are not included ,i have installed them separately using
opkg manager ,its working fine now..

gotthardp, 2018-06-20:
I see. Thank you for the hint.