gotthardp/lorawan-server #349: compiling lorawan-server on embedded linux

ennasiry, 2018-03-28:
hello,
Im trying to make lorawan-server using bitbake ,i get an error with the following log :
/usr/bin/env: ‘escript’: No such file or directory
any ideas how to solve it ?
( erlang already exist in my sysroots files )

gotthardp, 2018-03-28:
What OS are you using as a build environment? The "escript" is part of Erlang, so I guess you didn't install some package.

ennasiry, 2018-03-28:
Im using linux ubuntu 16.04 LTS ,when i run : $ which escript i get
/usr/bin/escript ,when i try to build my recipe using bitbake i get that
error
log :
ERROR: lorawan-server-0.5.3-r0 do_compile: oe_runmake failed
ERROR: lorawan-server-0.5.3-r0 do_compile: Function failed: do_compile (log
file is located at
/home/iotserver/oe-tra/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/lorawan-server/0.5.3-r0/temp/log.do_compile.21985)
ERROR: Logfile of failure stored in:
/home/iotserver/oe-tra/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/lorawan-server/0.5.3-r0/temp/log.do_compile.21985
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 release
*| /usr/bin/env: ‘escript’: No such file or directory*
| Makefile:35: recipe for target 'release' failed
| make: *** [release] Error 127
| ERROR: oe_runmake failed
| WARNING:
/home/iotserver/oe-tra/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/lorawan-server/0.5.3-r0/temp/run.do_compile.21985:1
exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at
/home/iotserver/oe-tra/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/lorawan-server/0.5.3-r0/temp/log.do_compile.21985)
ERROR: Task
(/home/iotserver/oe-tra/build/../layers/meta-lorawan/recipes-core/lorawan-server/lorawan-server_0.5.3.bb:do_compile)
failed with exit code '1'

gotthardp, 2018-03-28:
If you are using standard 16.04 this will not work anyway as https://packages.ubuntu.com show that you have Erlang v18 only. The Erlang v19 is required.

Could you please try following these steps: https://tecadmin.net/install-erlang-on-ubuntu and install a newer Erlang on your machine?

ennasiry, 2018-03-28:
> erlang-on-ubuntu and install a newer Erlang on your machine?

Im using LTS 16.04 and i have erlang 20 and nott 18 !
$ erl -v
Erlang/OTP 20 [erts-9.3] [source] [64-bit] [smp:4:4] [ds:4:4:10]
[async-threads:10] [hipe] [kernel-poll:false]

gotthardp, 2018-03-28:
Well, if you have https://packages.ubuntu.com/artful/erlang-base installed, there should be the `/usr/lib/erlang/bin/escript` on your machine. Is it there?

ennasiry, 2018-03-28:
yes it is there !

gotthardp, 2018-03-28:
And what happens when you type simply `escript`? The search PATH seems to be broken somehow.

ennasiry, 2018-03-28:
when i write escript i get :
escript: Missing filename
so the problem is related to my host ? not to lorawan-server ?

ennasiry, 2018-03-28:
my path is like this :
 echo $PATH
/home/iotserver/oe-tra/layers/openembedded-core/scripts:/home/iotserver/oe-tra/layers/openembedded-core/bitbake/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-8-oracle/bin:/usr/lib/jvm/java-8-oracle/db/bin:/usr/lib/jvm/java-8-oracle/jre/bin

2018-03-28 17:00 GMT+01:00 NOUREDDINE ENNASIRY <
ennasiry.noureddine@gmail.com>:

> when i write escript i get :
> escript: Missing filename
> so the problem is related to my host ? not to lorawan-server ?
>
>
>> And what happens when you type simply escript? The search PATH seems to
>> be broken somehow.
>>
>>
>
>
>
>
> Cordialement
>
> EN-NASIRY  NOUREDDINE
>
>
> +212 6 27 01 56 91
>
> ennasiry.noureddine@gmail.com

gotthardp, 2018-03-28:
The problem is related to your build environment. Somehow.

ennasiry, 2018-03-29:
where can i find lorawan-server project *escript* executable ?

gotthardp, 2018-03-29:
It's part of the Erlang environment. Usually it's in /usr/lib/erlang/bin/escript, but bitbake will likely have it's own somewhere in the build tree.

ennasiry, 2018-03-29:
yes i cant find it there and i also find the link to it located in /usr/bin
,but i still have the same error :
*/usr/bin/env: ‘escript’: No such file or directory*
even that i have ~/bin in my path --
echo $PATH
/home/iotserver/oe-tra/layers/openembedded-core/scripts:/home/iotserver/oe-tra/layers/openembedded-core/bitbake/bin:/bin:/usr/bin

ennasiry, 2018-03-29:
do i have to install rebar3 on my building environment or erlang is enough
?

2018-03-29 12:24 GMT+01:00 NOUREDDINE ENNASIRY <
ennasiry.noureddine@gmail.com>:

> yes i cant find it there and i also find the link to it located in
> /usr/bin ,but i still have the same error :
> */usr/bin/env: ‘escript’: No such file or directory*
> even that i have ~/bin in my path --
> echo $PATH
> /home/iotserver/oe-tra/layers/openembedded-core/scripts:/
> home/iotserver/oe-tra/layers/openembedded-core/bitbake/bin:/bin:/usr/bin
>
>
>
>> It's part of the Erlang environment. Usually it's in
>> /usr/lib/erlang/bin/escript, but bitbake will likely have it's own
>> somewhere in the build tree.
>>
>>
>
>
>
>
> Cordialement
>
> EN-NASIRY  NOUREDDINE
>
>
> +212 6 27 01 56 91
>
> ennasiry.noureddine@gmail.com

gotthardp, 2018-03-29:
rebar3 is downloaded automatically when it's missing

closed 2018-03-30 by ennasiry