gotthardp/lorawan-server #350: Send Downlink and Class C LoRa

pisua, 2018-03-28:
Hi, i have a issue about the sending of downlink on class C device. 
I 've configure a class C device that listen continiously downlink but when i send to the gateway a downlink using http rest api, the downlink is only sent to the device when the gateway receive a uplink from it.
I've tried to add "immediately" property on the JSON send to the gateway to the downlink. 
Today i'm stuck and don't know what should be configure about the device or network to be able to send downlink when even if the device don't send a uplink. 
Thanks in advance for your answer

matiasz8, 2018-03-28:
H pisua,

Please, could you tell us what manufacturer are you using?
I'm using a semtech RHF76-052 module by Rising HF

pisua, 2018-03-28:
we are using the for the device the ATIM card ATIM ARM N8 LW. here the link of the manufacturer of th card 
http://www.atim.com/fr/produits/catalogue/modules-rf/module-carte-radio-lorawan-arm-n8-lw/

gotthardp, 2018-03-28:
Having the "immediately" property and a port number is a must. After you send the Class C downlink, does the message wait until the uplink is received? That sounds like you didn't set the "immediately" properly. Could you share with us the JSON you send?

pisua, 2018-03-28:
Ok, i've looked on another issue where you explain how to set the flag immediately and in my case i was imagined it should be passed like this in the json
{"data" :"ABCD" , "immediately":true}

For the port i don't know regarding my configuration which value to set. i assume the value should be passe like that with the immediately flag  ? : 
{"data":"ABCD" , "time":"immediately", "port": 0}. 

Maybe the link JSON Payload in that page https://github.com/gotthardp/lorawan-server/blob/208e38f82374c1db8e1d829c8ca3ba4a2a500236/doc/Introduction.md should be fix to be sure we understand how the JSON payload send to the gateway should be set and what are the must value. I've seen that it's not working.

pisua, 2018-03-28:
Yes, it works better but not until the end. i don't receive anything but the downlink is no longer pending.
 i got an error in the debug.log  link to my node :
 
2018-03-28 18:10:46.477 [debug] <0.20209.17>@lorawan_mac_commands:build_fopts:74 0124E785 <- [dev_status_req]
2018-03-28 18:10:46.485 [error] <0.313.0> node 0124E785 <<"tx_freq">>.

pisua, 2018-03-28:
ok i found it it was a misconfiguration in my network on the gateway. thanks for you help. And for the link JSON Payload it could be great to have it available again to see what kind of property we can set on downlink.

closed 2018-03-28 by pisua

gotthardp, 2018-03-28:
I am about to fix that link.

chansoonthiam, 2019-01-23:
i have question on the LoRaWAN Class C for lamps application. How many expected missing downlink command from LNS to node

BEAJousama, 2019-06-07:
Hi, 
Please how do you send downlink frames via REST API ? I didn't find a rest ressource where to post txframes 

Thanks in advance

gotthardp, 2019-06-07:
@BEAJousama, you have to create HTTP connector (and a handler). See https://github.com/gotthardp/lorawan-server/blob/master/doc/Connectors.md#httprest