gotthardp/lorawan-server #351: Server/Gateway to node downlink communication

gmbankar37, 2018-03-29:
How can send data from Server/Gateway to the node device( LoRa RAK811 module) connected to a RPi3? I read about it and got to know it should be send from the application server. But I don't see any such feature here. Please help me out. Sorry this is not an issue and I am just inquiring. Thank you.

gotthardp, 2018-03-29:
Hello. Start with https://github.com/gotthardp/lorawan-server/blob/master/doc/Introduction.md#downlink

gmbankar37, 2018-04-02:
> doc/Introduction.md#downlink

Hello,
Thank you. for the document provided. I did read the document provided by
you. I wanted to know that can I send a downlink signal from the lorawan
server without the backend server addition?

Gaurav Bankar

gotthardp, 2018-04-02:
What is your use-case? What do you want to be the trigger of the downlink?

gmbankar37, 2018-04-02:
My LoRa node is connected to a data monitoring device and I want to send
the data through gateway to a server where I can see the data. For now I
want to send data to node and see if it can receive it and I can see it. I
might want to send a query in future for retrieve a specific data. Also I
haven't done any coding in Erlang so its difficult to modify the already
existing code.
So If I have to develop external application according to my requirement do
I have to write entirely different Erlang code ?

gotthardp, 2018-04-02:
For such tests I'd recommend to use the web socket demo application (see the Connectors guide).

gmbankar37, 2018-04-02:
Hey how do I change the band? I changed the network from EU868 tp IN865 but
the node still shows EU868.

gotthardp, 2018-04-02:
Where does the node show EU868?
The firmware is region specific, i.e. you cannot switch the region as recognized by the end-device from the server side. You need to reconfigure/change your firmware.

gmbankar37, 2018-04-03:
Hello,
 I am trying to have a websocket server on the lorawan-server. Do I have to
add the demo client page anywhere inside the source file of the
lorawan-server ?
 when I enter this URL ws://server:8080/ws/uplink/<DevAddr> in browser it
shows error as page not found or site cant be reached according to the
screen shot attached.

Thank you,
Gaurav Bankar

gotthardp, 2018-04-03:
https://github.com/gotthardp/lorawan-server/blob/master/doc/Connectors.md#demo-page

gmbankar37, 2018-04-03:
> doc/Connectors.md#demo-page

I have followed the same link/document you gave me. When I complete
everything setting up the handler and connector I am not able to connect to
the websocket. Also the demo app gives error. Please look at the
screenshots to see if I have made any mistake during setup.

gotthardp, 2018-04-03:
I don't see any screenshots.

gmbankar37, 2018-04-03:
[image]
[image]

gmbankar37, 2018-04-03:
[image]

gotthardp, 2018-04-03:
Sir, the instructions I sent you say you should enter `http://127.0.0.1:8080/admin/ws.html` and put the `/ws/events/...` URI there.

gmbankar37, 2018-04-03:
The lorawan-server as websocket says The *Subscribe* and *Received Topic*
fields are not used and can be left empty. But if the received topic is
empty I will not be able to send downlink message. So do I have to make
changes in the demo client or the handler to make it happen?

gotthardp, 2018-04-03:
For WS there is only one URI: you connect to it, receive uplinks and send downlinks. The MQTT works differntly so it needs also the sub/rec topics; WS doesn't.

gmbankar37, 2018-04-04:
Hi,
  I was able to send data from demo client.  As my lorawan-server acts as
Websocket it should receive the data. But I am not able to see the data in
the received frames. So the data I am sending through demo client is that
downlink or uplink? Thank you.

gmbankar37, 2018-04-05:
> The data you sent is a "downlink" and does not appear in the Received
> Frames list.

So where can I see this data ? In the downlink of activated nodes ?

gotthardp, 2018-04-05:
For class A, before sent, messages are listed on the node page. Otherwise the downlinks are not stored.

gotthardp, 2018-04-05:
The request to long and display downlinks is now Issue #362.
I believe the other questions were clarified, so I am closing this.

closed 2018-04-05 by gotthardp