gotthardp/lorawan-server #352: Unexpected mtype 3

psgenn, 2018-04-01:
Hi, I'm trying to send confirmation messages and the server gives an error "Unexpected mtype 3". What am I doing wrong?

`Apr 01 09:13:10 raspberrypi lorawan-server[13150]: 09:13:10.249 [error] <<"11111111">> Unexpected mtype 3
`

[image]
[image]

gotthardp, 2018-04-01:
How do you send the mtype 3 messages?

psgenn, 2018-04-01:
I do not do anything for this. I'm using the firmware version from the http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32cube-expansion-packages/i-cube-lrwan.html

gotthardp, 2018-04-02:
MType=3 is "unconfirmed downlink". This is sent by the server, so if your device sends MType=3, it is apparently malfunctioning.

closed 2018-04-19 by gotthardp