gotthardp/lorawan-server #353: Info on erlang fun

webfrank, 2018-04-03:
Hi,
   I'm trying to connect NKE Watteco end devices, one Temp/Hum and one Current. They use a proprietary ZCL protocol and I need to decode the incoming message based on some previous decode byte.
   Could you provide with some sample?

gotthardp, 2018-04-03:
Hi. Some examples are here:
https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md#parse-uplink

webfrank, 2018-04-03:
Hi, yes I know the basics of parsing binary data but I would like to implement some logic like if valueA==0x8A parse this way otherwise parse the other way. Is it possible?

gotthardp, 2018-04-03:
It's always possible, but ZCL is a bit complex, so writing a fun might not be the most convenient option. An alternative would be to add ZCL to the supported protocols, if possible, and implement it natively inside the server just like Cayenne.

webfrank, 2018-04-03:
Knowing how to do it :) I think I will decode data server side.

gotthardp, 2018-04-03:
And don't forget that contributions are always welcome ;-)

closed 2018-04-03 by gotthardp