gotthardp/lorawan-server #355: Error on uplink

webfrank, 2018-04-03:
2018-04-03 09:32:20.013 [error] <0.406.0>@lorawan_mac_region:dr_to_down:102 gen_statem <0.406.0> in state uplink terminated with reason: bad arithmetic expression in lorawan_mac_region:dr_to_down/3 line 102
2018-04-03 09:32:20.085 [error] <0.406.0>@lorawan_mac_region:dr_to_down:102 CRASH REPORT Process <0.406.0> with 0 neighbours crashed with reason: bad arithmetic expression in lorawan_mac_region:dr_to_down/3 line 102
2018-04-03 09:32:20.085 [error] <0.313.0> Supervisor lorawan_handler_sup had child handler started with lorawan_handler:start_link() at <0.406.0> exit with reason badarith in context child_terminated

gotthardp, 2018-04-03:
Could you post a screenshot of your Node's ADR settings?

webfrank, 2018-04-03:
[image]

gotthardp, 2018-04-03:
This looks like 3 successful uplinks. I don't see any problem there.

webfrank, 2018-04-03:
I managed to send uplink but the server reports this error, I don't know why

webfrank, 2018-04-03:
Something strange now, I could only receive first uplink over mqtt, subsequent data, received by server, is not forwarded to mqtt

gotthardp, 2018-04-03:
Would you share also the crash.log, please?

webfrank, 2018-04-03:
2018-04-03 12:49:28 =ERROR REPORT====
** State machine <0.8193.0> terminating
** Last event = {cast,{rxq,[{<<114,118,255,0,57,3,0,254>>,{rxq,868.1,<<"SF12BW125">>,<<"4/5">>,undefined,120975916,-28,8.8},undefined}]}}
** When server state  = {uplink,{-576448885261,{{network,<<"SAC">>,<<0,0,0>>,undefined,<<"EU868">>,<<"4/5">>,5,6,1,2,16,14,10,1,0,{0,0,869.525},[{0,7}],undefined},{profile,<<"WATTECO">>,<<"SAC">>,<<"Analog">>,<<"Analog">>,true,2,0,0,{undefined,undefined,undefined},undefined,{undefined,undefined,868},true},{node,<<1,20,194,161>>,<<"WATTECO">>,<<"TEMP">>,<<59,16,123,116,89,199,9,233,167,248,152,155,133,251,72,114>>,<<90,5,231,143,58,82,162,231,62,85,213,227,166,84,90,107>>,undefined,7,1,{{2018,4,3},{12,42,19}},{{2018,4,3},{12,42,19}},0,{{2018,4,3},{12,48,28}},[{<<114,118,255,0,57,3,0,254>>,{rxq,868.5,<<"SF12BW125">>,<<"4/5">>,undefined,60976788,-27,6.8}}],1,undefined,{10,0,[{0,7}]},[],{0,0,869.525},[],[{-27,6.8},{-29,7.2},{-17,10.0},{-17,9.5},{-26,8.0},{-20,10.0},{-25,9.0}],undefined,{{2018,4,3},{12,43,28}},1,[{{{2018,4,3},{12,43,28}},254,13,-20.0}]}},{frame,1,<<1,20,194,161>>,1,0,0,7,<<5,7>>,125,<<17,10,4,2,0,0,41,8,168>>},{{handler,<<"Analog">>,[<<"devaddr">>,<<"app">>,<<"appargs">>,<<"data">>,<<"datetime">>,<<"rssi">>,<<"port">>,<<"lsnr">>,<<"deveui">>,<<"battery">>],undefined,undefined,[<<"app">>,<<"event">>,<<"devaddr">>,<<"deveui">>,<<"appargs">>,<<"datetime">>],{<<"fun(Vars, Event) ->\n  Vars#{event => Event}\nend.">>,#Fun<erl_eval.12.118419387>},undefined,<<"never">>},#{app => <<"Analog">>,appargs => <<"TEMP">>,battery => 254,data => <<17,10,4,2,0,0,41,8,168>>,datetime => {{2018,4,3},{12,49,28}},devaddr => <<1,20,194,161>>,deveui => <<112,179,213,231,94,0,76,82>>,port => 125}}}}
** Reason for termination = error:badarith
** Callback mode = state_functions
** Stacktrace =
**  [{lorawan_mac_region,dr_to_down,3,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,102}]},{lorawan_mac_region,datar_to_down,3,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,89}]},{lorawan_mac_region,tx_offset,4,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,82}]},{lorawan_mac_region,rx1_window,3,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,26}]},{lorawan_handler,uplink,3,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_handler.erl"},{line,151}]},{gen_statem,call_state_function,5,[{file,"gen_statem.erl"},{line,1223}]},{gen_statem,loop_event,6,[{file,"gen_statem.erl"},{line,995}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]
2018-04-03 12:49:28 =CRASH REPORT====
  crasher:
    initial call: lorawan_handler:init/1
    pid: <0.8193.0>
    registered_name: []
    exception error: {badarith,[{lorawan_mac_region,dr_to_down,3,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,102}]},{lorawan_mac_region,datar_to_down,3,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,89}]},{lorawan_mac_region,tx_offset,4,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,82}]},{lorawan_mac_region,rx1_window,3,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,26}]},{lorawan_handler,uplink,3,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_handler.erl"},{line,151}]},{gen_statem,call_state_function,5,[{file,"gen_statem.erl"},{line,1223}]},{gen_statem,loop_event,6,[{file,"gen_statem.erl"},{line,995}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
    ancestors: [lorawan_handler_sup,<0.310.0>,<0.307.0>,<0.278.0>]
    messages: []
    links: [<0.313.0>]
    dictionary: []
    trap_exit: false
    status: running
    heap_size: 4185
    stack_size: 27
    reductions: 2126
  neighbours:
2018-04-03 12:49:28 =SUPERVISOR REPORT====
     Supervisor: {local,lorawan_handler_sup}
     Context:    child_terminated
     Reason:     badarith
     Offender:   [{pid,<0.8193.0>},{id,handler},{mfargs,{lorawan_handler,start_link,[]}},{restart_type,transient},{shutdown,5000},{child_type,worker}]

gotthardp, 2018-04-03:
It is "undefined" in the profile as indicated by the error report above
`{profile,<<"WATTECO">>,<<"SAC">>,<<"Analog">>,<<"Analog">>,true,2,0,0,{undefined,undefined,undefined},undefined,{undefined,undefined,868},true}`
but from the code I don't understand how an undefined offset in the profile could cause this error.

altishchenko, 2018-04-03:
@gotthardp How is the nodes rx_win tuple built? The offset in question comes from it, by the chain of funcs.

altishchenko, 2018-04-03:
@gotthardp On the other hand, remember @webfrank had an issue with deleting/changing profile on the node. It looks like the starting point of events.

webfrank, 2018-04-04:
These are the profile screenshots. Setting RX1 DR offset to 0 resolved the error. Should it be 0 by default?

I also noticed that even I have set ADR off the Node has ADR on. I deleted node and commissioned records and recreated them but ADR is still on although I set off on profile.
Does the server "remembers" previous configuration?

[image]
[image]

altishchenko, 2018-04-04:
@webfrank I can see a small discrepancy here - your device states that it uses ADR and 8 channels, but profile states that ADR is disabled and default channels are used. Could it be that this mismatch was introduced when you deleted/re-created the profile? @gotthardp ?

altishchenko, 2018-04-04:
Oops, didn't read the text first. This is wierd, it definitely should be disabled on the node...

altishchenko, 2018-04-04:
@webfrank what is in your Network configuration?

altishchenko, 2018-04-04:
@webfrank wierd, looks like Network settings take precedence over Profile

altishchenko, 2018-04-17:
@webfrank is this solved?

gotthardp, 2018-04-17:
But I did reproduce this bug. I will keep this open until I find and fix it.

romansoft, 2018-05-02:
I recently started getting lots of uplink errors with similar error messages related to bad arithmetic, but I doubt my problem is caused by the same missing parameter issue. I started seeing problems right after connecting a new gateway the model I never tried before. Here are excerpts from the log files (just for one error example). Not sure what all these "undefined" values are all about (the profiles are complete), but this value looks very strange in particular: <<"¹yM\vX{»°H7]X?öåÜ">>

Looks like the NS state got corrupted since the affected nodes are not able to make successful uplinks after removing the suspect gateway and using a good, known gateway until they are rejoined. Is it possible the gateway corrupted the NS or is this is some other issue? Any way to restore nodes to good state without forcing them to rejoin (the physical process is cumbersome on field nodes...)?

**debug.log**
2018-05-02 03:29:30.188 [error] <0.1270.0>@lorawan_mac_commands:handle_fopts:19 gen_statem <0.1270.0> in state uplink terminated with reason: no match of right hand value {aborted,{badarith,[{lists,sum,2,[{file,"lists.erl"},{line,291}]},{lorawan_mac_commands,average_qs0,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,64}]},{lorawan_mac_commands,average_qs,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,61}]},{lorawan_mac_commands,append_qs,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default..."},...]},...]}} in lorawan_mac_commands:handle_fopts/4 line 19
2018-05-02 03:29:30.189 [error] <0.1270.0>@lorawan_mac_commands:handle_fopts:19 CRASH REPORT Process <0.1270.0> with 0 neighbours crashed with reason: no match of right hand value {aborted,{badarith,[{lists,sum,2,[{file,"lists.erl"},{line,291}]},{lorawan_mac_commands,average_qs0,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,64}]},{lorawan_mac_commands,average_qs,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,61}]},{lorawan_mac_commands,append_qs,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default..."},...]},...]}} in lorawan_mac_commands:handle_fopts/4 line 19
2018-05-02 03:29:30.189 [error] <0.1269.0>@lorawan_mac_commands:handle_fopts:19 Supervisor lorawan_handler_sup had child handler started with lorawan_handler:start_link() at <0.1270.0> exit with reason no match of right hand value {aborted,{badarith,[{lists,sum,2,[{file,"lists.erl"},{line,291}]},{lorawan_mac_commands,average_qs0,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,64}]},{lorawan_mac_commands,average_qs,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,61}]},{lorawan_mac_commands,append_qs,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default..."},...]},...]}} in lorawan_mac_commands:handle_fopts/4 line 19 in context child_terminated

**crash.log**
2018-05-02 03:29:30 =ERROR REPORT====
** State machine <0.1270.0> terminating
** Last event = {cast,{rxq,[{<<0,128,0,0,160,0,21,110>>,{rxq,903.3,<<"SF8BW125">>,<<"4/5">>,undefined,2371067908,-93,7.8},undefined},{<<100,127,218,255,254,0,82,145>>,{rxq,903.3,<<"SF8BW125">>,<<"4/5">>,{{2018,5,2},{3,29,29.0}},1086980060,-95,11.2},undefined}]}}
** When server state  = {uplink,{-576456815480,{{network,<<"NA_Network_8ch">>,<<0,0,1>>,undefined,<<"US902">>,<<"4/5">>,5,6,1,2,26,30,5,5,4,{0,8,923.3},[{0,7},{64,64}],undefined},{profile,<<"SMP">>,<<"NA_Network_8ch">>,<<"Soil Moisture">>,<<"Inet">>,true,1,0,1,{5,1,[{0,7},{64,64}]},2,{0,8,923.3},true},{node,<<2,134,56,10>>,<<"SMP">>,<<"SN-200016">>,<<128,169,227,125,177,95,63,145,212,139,184,243,254,66,241,115>>,<<"¹yM\vX{»°H7]X?öåÜ">>,107,10,{{2018,4,28},{6,6,37}},{{2018,4,28},{6,6,37}},0,{{2018,5,1},{20,28,37}},[{<<100,127,218,255,254,0,82,145>>,{rxq,902.3,<<"SF8BW125">>,<<"4/5">>,{{2018,5,1},{20,28,36.0}},3464104516,-76,10.5}}],1,undefined,{5,2,[{0,7},{64,64}]},[],{0,8,923.3},[],[{-76,10.5},{-75,8.8},{-71,10.5},{-71,11.0},{-75,10.5},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{undefined,undefined},{-70,10.5},{-76,10.8},{-76,8.0},{-76,10.0},{-75,11.2},{-77,11.8},{-75,10.8},{-75,8.5},{-75,11.5},{-75,10.5},{-81,8.2},{-91,3.0},{-83,10.2},{-89,11.5},{-85,7.5},{-85,8.0},{-92,10.8},{-89,7.2},{-87,8.2},{-89,6.5},{-89,10.8},{-91,4.5},{-86,8.2},{-88,9.2},{-91,8.0},{-85,9.5},{-95,0.0},{-81,10.8},{-77,11.5},{-76,11.0},{-76,11.5},{-75,11.5},{-93,8.0}],undefined,{{2018,5,1},{14,27,51}},94,[{{{2018,5,1},{14,27,51}},168,-3,-10.0},{{{2018,4,30},{23,25,59}},168,7,-10.0},{{{2018,4,30},{9,37,35}},168,5,-10.0},{{{2018,4,29},{20,35,58}},168,7,-10.0},{{{2018,4,29},{8,34,16}},167,6,-15.0},{{{2018,4,28},{19,32,22}},168,6,-15.0},{{{2018,4,28},{6,6,57}},167,8,-15.0}]}},{frame,0,<<2,134,56,10>>,1,0,0,107,<<>>,6,<<0,0,0,0,0,0,0,0,0,0>>},{{handler,<<"Soil Moisture">>,[<<"app">>,<<"deveui">>,<<"fcnt">>,<<"port">>,<<"data">>,<<"datetime">>,<<"datr">>,<<"freq">>,<<"mac">>,<<"lsnr">>,<<"rssi">>,<<"appargs">>,<<"battery">>,<<"devaddr">>],undefined [... cut ...] awan_handler,uplink,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_handler.erl"},{line,130}]},{gen_statem,call_state_function,5,[{file,"gen_statem.erl"},{line,1240}]},{gen_statem,loop_event,6,[{file,"gen_statem.erl"},{line,1012}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
     Offender:   [{pid,<0.1270.0>},{id,handler},{mfargs,{lorawan_handler,start_link,[]}},{restart_type,transient},{shutdown,5000},{child_type,worker}]

gotthardp, 2018-05-02:
It feels like the gateway is sending bad/no RSSI/SNR values. Do you have a log from the gateway, please?

romansoft, 2018-05-02:
I'm not sure how to view logs on this gateway yet. I'll ask the manufacturer (Tektelic). I tried a different version of their provisioning tool and it seems it's not corrupting node state anymore, but the U/L RSSI and SNR are not being reported on the Activated (Nodes) screen for a node:
[image]

Here are some screen grabs from Wireshark showing the U/L payloads:
[image]

[image]
 
SNR and RSSI seem to be reported in JSON.

romansoft, 2018-05-02:
Looks like the format and field naming in the JSON from PF is quite different for Tektelic Kona versus Multitech Conduit. Perhaps this is causing these issues?

**Tektelic:**
[image]

**Conduit:**
[image]

gotthardp, 2018-05-02:
That is definitely causing the problems. It violates the standard protocol specification https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT
Do you know why they did that? Is there a larger group of gateways that support this version of the protocol?
If they are agreeable, one solution could be to include the rssi and lsnr twice: once in the standard way and once in their way.

gotthardp, 2018-05-02:
It seems to be in Kerlink too https://github.com/brocaar/lora-gateway-bridge/issues/44

romansoft, 2018-05-02:
No idea why. This gateway is supporting 16 channels (and another model supporting 64 channels), so perhaps this is why they have rsig array, one for each 8-ch subband. 

Tektelic is gaining a lot of traction (at least in US), so it would be important to support it.

I will talk to them about it. Could you please propose the object format that you could support? 
If they are not agreeable, what are the options to work around this in the short term?

romansoft, 2018-05-02:
Thanks. Here is the screen grab for the downlink

[image]

gotthardp, 2018-05-02:
That's the gateway stats. The downlink starts with `txpk`. It's sent by the server, so you'd need to find a server that supports this extension.

romansoft, 2018-05-02:
According to the comments at https://github.com/brocaar/lora-gateway-bridge/issues/44, it looks like Kerlink and Tektelic are using older version of PF (v1.0) that references the structure we've been seeing. https://github.com/brocaar/lora-gateway-bridge/files/1128419/PROTOCOL.TXT
If this is the case, it will be hard to convince Tektelic to make any changes, as they will claim compliance (albeit with older version of protocol).

gotthardp, 2018-05-02:
The comment says "the protocol.txt included in the package". If you open the V1.0 of the protocol as available on the GitHub, you get the standard protocol: https://github.com/Lora-net/packet_forwarder/blob/v1.0.0/basic_pkt_fwt/PROTOCOL.TXT
It's either some ancient protocol (prior it was stored in GitHub), or still proprietary changes reflected in the protocol specification. It would be great if you confirm with the vendor what kind of protocol they implemented ;-)

romansoft, 2018-05-02:
Yes, I'll try to talk to them today.

romansoft, 2018-05-02:
I talked to Tektelic. This format is indeed due to multi lora transceiver
support. Semtech defined this initially for this purpose. The protocol is
as was listed in the brocaar's URL for v1.0 and will be the same for
Tektelic and Kerlink.

It will be really good if you could support this format as these multi
transceiver gateways are becoming less expensive and more popular.

gotthardp, 2018-05-04:
@romansoft I hope I implemented that. Try the latest master HEAD.

romansoft, 2018-05-04:
that was quick, @gotthardp . Thanks!
Yes, I tried this and it works fine.

However, I found another issue: when a gateway is deleted from NS, it is still accepting uplinks from nodes. This is unexpected and undesired behavior.

gotthardp, 2018-05-04:
OK, cool. For the second problem I creates a new issue #383.

gotthardp, 2018-08-10:
I assume the error was (finally) found and fixed.

closed 2018-08-10 by gotthardp