gotthardp/lorawan-server #356: Error on downlink

webfrank, 2018-04-03:
If I send a new downlink to node via MQTT I have three messages queued

gotthardp, 2018-04-03:
Then you hit the button 3x, or there were 2 messages aready queued. Please be specific about the issue and provide enough information for debugging.

webfrank, 2018-04-03:
I sent only one message using MQTT but i found three messages queued

[image]

gotthardp, 2018-04-03:
Could you have published 3 messages accidentally? Other people don't seem to have this problem.

webfrank, 2018-04-03:
No, every downlink I send gets multiplied :(

altishchenko, 2018-04-03:
Sorry to interrupt, my two cents:
How many devices do you have?
What are the topics settings on the connector?

webfrank, 2018-04-03:
Yes, I have 3 mqtt connectors as I cannot link 3 handlers to the same connector (see previous issue)

altishchenko, 2018-04-03:
Then, there are two screenshots. Handler and typical connector.

webfrank, 2018-04-04:
Here are the screenshots but the problem is that I want to use the same connector for several handlers but as I cannot link multiple handlers to one connector I had to duplicate connectors. This is why I got 3 messages queued, one for every connector configured.

[image]
[image]
[image]
[image]

gotthardp, 2018-04-04:
So you connected 3 connectors to the same handler and subscribed each of the connector to the same MQTT topic? In such case it is expected you will get 3 messages, isn't it?

altishchenko, 2018-04-04:
@webfrank the way you configured your connector is fine, when it is single on your server. If have multiple connectors configured the same, then each of them will get a message back, the received topic filter will work correctly on each of them resulting in a correct message for the existing device, hence your multiplication. You should subscribe to 'xxx/{devaddr}/downlink' in such situation.

webfrank, 2018-04-04:
Ok but doesn't would be better to have multiple handlers for one connector? I changed the configuration of topics prefixing with app name and now I have messages queued correctly.

altishchenko, 2018-04-04:
No no, you got it wrong a bit. The way you configured you connector **is the way** for multiple handlers-OneConnector scenario, this is how it was visioned and implemented. You stated that you have three similar connectors - hence you have to specific on each one of them. If your issue with uplink is somewhat cured, try going back to a single connector. Don't forget to change the handlers first :)

altishchenko, 2018-04-04:
@webfrank I didn't migrate from 0.4 in a production system yet and 0.5 is running on a test bed with a small number of devices... The behaviour you want **was the behaviour** of 0.4.13. I am not sure why Petr has changed that. I just checked with the server and I believe it is a stopping problem for my future migration too... @gotthardp Petr can you comment on that?

gotthardp, 2018-04-04:
I'm afraid that for now a Connector can be assigned to one Application/Handler only. This was done to limit the complexity of the templated connections (one MQTT connection/subscription per device). I thought that creating one connector for each Handler is not a big issue compared to the nightmare of implementing templated connections from multiple handlers.

altishchenko, 2018-04-04:
@gotthardp : Petr, I understand that - I scanned the new code and it is architectural = means no easy way out, but the real problem lies elsewhere. Look, I have this setup with 0.4.13+ at the moment: there is a number of servers running, there is quite a number of different types of devices with different binary data transmitted, but all of that is handled on a central broker cluster that notifies different **applications** that decide their actions next. Unfortunately, this is a mine (uranium mine for that) and a life and death situation, so quite a number of vendor-made apps are sitting on the broker and deciding if they should raise the ambulance or fire brigade and such. Having a separate connector for **each** special **type** of the device is not that handy - this is what handlers are for, running 100+ parallel similar connections from a server to the broker cluster is not a very good situation either. Esp, in the middle of Taiga... :) (The sensors/motes we are talking about are everything from fall detection to radiation meters + gps/glonass positioning, on the head hit and strength counter, etc..) Most of them are embedded into the workers helmets.

gotthardp, 2018-04-05:
The request to have multiple handlers connected to one connector is Issue #339.
The other points were clarified, so I am closing this.

closed 2018-04-05 by gotthardp