gotthardp/lorawan-server #357: Lots of ack_lost 100.0 warnings for high latency network

romansoft, 2018-04-03:
I'm seeing lots of ack_lost 100.0 warnings when using cellular gateway that has high network latency. 

[image]

The gateway status window shows about 210ms network delay. The gateway's (Multitech Conduit) Packet Forwarder was configured with a larger timeout ("push_timeout_ms":300). 

I don't suspect there is any issue with the network server implementation. I'm just trying to understand how to fine-tune the parameters in such situations to make the system work better and to understand the limits. Perhaps the insights will be useful to others as well.

So, my questions:
1. When the gateway status reports 210ms network delay, what exactly does this mean? Is it the round-trip or gateway-to-server delay?
2. What is exactly the definition of push_timeout_ms? Looking at SPF source code, it looks this is divided by 2 before programming socket's RX timeout. So, wouldn't this be the max expected roundtrip latency (or max expected latency for server-to-gateway times 2)?
3. Would I need to set push_timeout_ms to >420 to avoid the ack_lost warnings?
4. What is the impact of missing push acknowledges on the system (ack_lost 100.0). Will it contribute to any issues uplinks or only a problem for downlinks?
5. What would be the maximum round trip network delay that the system could tolerate for a) uplinks and b) downlinks?

Any insights would be very helpful.

gotthardp, 2018-04-03:
That really a lot ack_losts.
To give a baseline: we are talking about this protocol: https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT
There is one packet_forwarder issue: "false" ack_lost is announced when the uplink is sent just before the reporting period, because the acknowledgement comes in the next reporting period. However, with this amount of errors this is probably not your case.
0) ack_lost means that PULL_ACK or PUSH_ACK were not received. It's just a warning, but it could be a symptom of some networking issue.
1) network delay is a round-trip of the PULL_RESP-TX_ACK
2) I don't know to be honest.
3) Try it.
4) It depends on what is the cause: the system may be fully functioning (if only the ACK was lost), or not (if the PULL_DATA, PUSH_DATA were actually lost).
5) If you update the timers it can survive even a high delay. Note there are two hidden xxx_delay config values in the server: https://github.com/gotthardp/lorawan-server/blob/master/src/lorawan_server.app.src

Just a question: I saw packet_forwarders using one port for uplinks and another port for downlinks. This is not supported by my server. I assume only a single UDP port used for all gw-srv communication.

romansoft, 2018-04-04:
Thanks for clarifications.

I have not had a chance to experiment with even longer push_timeout_ms setting yet. I'll try as soon as this gateway is available to me again.

gateway_delay seems to be used for calculating which RX window should be used, does it? I guess it should not affect PUSH_ACK, it it might affect the timing of PULL_ACK. Yes?
Is it exposed through sys.config or would it require code recompilation to change it? 

I configured packet forwarder with the same port for both uplinks and downlinks, as in the documentation.

gotthardp, 2018-04-04:
The hidden config values can be added to sys.config anf changed, no recompilation is needed.

Yes, gateway_delay is used to auto-select RX window (so the server knows if it can still catch the RX1 or if it's too late).

The PULL_ACK is sent immediately after receiving PULL_DATA. The gateway_delay config parameter itself has no impact. However, very long delay on the gw-to-srv connection may cause a pull-ack-timeout, but it depends on the packet_forwarder implementation, which I don't know.

romansoft, 2018-04-10:
I've tried again with higher push_timeout and gateway_delay values. Nothing changed. The UL/DL functionality seems OK despite numerous warnings, which clutter the dashboard. Not sure what else to try. Is there an easy way to determine if the warnings come from PUSH_ACK or PULL_ACK activity?

On a somewhat related topic, what is the effect of increasing stat_interval on the server? The concern is for cellular gateways to reduce the amount of data it sends to the server to reduce the cellular data plan cost. What could be the max value? without causing adverse effects?

gotthardp, 2018-04-10:
The best way to debug this is to take a Wireshark trace of the server-to-gateway communication and see if the packet was sent/received and when.

The stat_interval is just number and it affects just how often the statistics are generated. You can set a very high number (I think that even 4290000).

gotthardp, 2018-04-12:
the rate of ack-lost and similar infos will be impacted by a longer stat-interval of the gateway

gotthardp, 2019-03-16:
I just added a section on ack_lost to the Troubleshooting guide.

closed 2019-03-16 by gotthardp