gotthardp/lorawan-server #360: device setting problem(by API) - application setting

nuguri618, 2018-04-05:
Hi,

I try to create device or node on LoRaWAN network server by API.
But Still how to add application by API.

I post new device by postman with this json body

[{'deveui': '100000000000EE04','region':'US902-928','application':'backend',
    'appeui':'0000000000000003', 'appkey':'92DF10F2C45208F050DA785D70075DE6'}]
But only application can't set up on network server.
is it possible by original API? or need some other thing?

gotthardp, 2018-04-05:
Everything you can do via web-admin can be done via API too.

nuguri618, 2018-04-05:
my json is incorrect?.. only application can't setting...

gotthardp, 2018-04-05:
Which server version are you using?

nuguri618, 2018-04-06:
I use version 0.4.13..

gotthardp, 2018-04-06:
What exactly do you want to achieve? What are you doing and how does the system react?

nuguri618, 2018-04-06:
I want to register device by API. but only application cannot setting by API.
I use this Json.
[{'deveui': '100000000000EE03','region':'US902-928','application':'backend',
'appeui':'0000000000000003', 'appkey':'92DF10F2C45208F050DA785D70075DE6'}]

and check on web.
[image]

nuguri618, 2018-04-06:
Thanks.. I got it! It works!

closed 2018-04-06 by nuguri618