gotthardp/lorawan-server #361: FR: access list to allow only certain networks to be processed

altishchenko, 2018-04-05:
This is the opposite of the 'Ignored' list, that allows only devices with the certain list of NetIds to be processed and displayed on the dashboard run scale. Blocked devices should be mentioned with the debug level only.
Reasoning behind: a party of LoRa enabled ABP devices (200+) in a crowded radio area. For the operator to be able to keep track of the events and everything else dashboard, received frames, events list should be limited to the networks it operates. Devices not belonging to the list of allowed networks should be mentioned at debug level only.

gotthardp, 2018-05-18:
Hello. In the new server release (0.5.4) the web-admin has a new Server flag called "Log Ignored". If this is set to "true" the unknown frames are not logged and do not appear anywhere. Would this solve your issue?

altishchenko, 2018-05-20:
Hello Petr, I am not sure if this is exactly the case for everyone, but this is probably true for the case where you want to see only your registered devices on the graph and in the log. Given that you know your DevEUIs or DevAddrs in advance this is a fix for the time.

closed 2018-05-20 by altishchenko