gotthardp/lorawan-server #363: OTAA : Question

joffrey07, 2018-04-10:
Hello,

I am new in LORA and I need some help to add new device to my LORA network using OTAA. 
When I want to add one node in Commisionned menu, I write the device's DevEUI, choose profile and put the appkey. 

But, when the node reach the network I have the BAD_MIC error. I don't know how determine the good appkey to add my device. I read the doc, do some test and I have not the solution.

Could you help me please.

Thank you by advance for your help and have a nice day.

gotthardp, 2018-04-10:
The AppKey configured in the device and the AppKey configured in the server must be identical. Usually the AppKey is configured during device manufacturing, so you should ask the device manufacturer about the AppKey and then configure it in the server.

joffrey07, 2018-04-11:
Thank you for these informations about AppKey.

Have a nice day.

closed 2018-04-11 by joffrey07