gotthardp/lorawan-server #364: NPM Issue with the make release

Gauss-Monitoring, 2018-04-11:
I KNOW IT'S NO A BIG ISSUE BUT COULD NOT FIND HOW TO SOLVE THE PROBLEM, ASK TO NPM, STILL WAITING FOR THE ANSWER 

root@Gauss:/lorawan-server# make release
===> Verifying dependencies...
===> Compiling lorawan_server
npm WARN deprecated angular-ui-router@0.4.3: This npm package 'angular-ui-router' has been renamed to '@uirouter/angularjs'. Please update your package.json. See https://ui-router.github.io/blog/uirouter-scoped-packages/
npm WARN deprecated coffee-script@1.10.0: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.
npm ERR! code 128
npm ERR! Command failed: /usr/bin/git clone -q git://github.com/nmccready/google-maps-utility-library-v3-markerwithlabel.git /root/.npm/_cacache/tmp/git-clone-b0ae8ff8
npm ERR! fatal: could not create leading directories of '/root/.npm/_cacache/tmp/git-clone-b0ae8ff8': Permission denied
npm ERR!

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-04-11T13_03_27_464Z-debug.log
===> Hook for compile failed!

altishchenko, 2018-04-11:
It is a known issue with the new npm. You have to downgrade npm and the best way is to downgrade it with npm :)
`sudo npm install -g npm@7.6.0` for example.

altishchenko, 2018-04-11:
I am still running at npm 5.4.2 without any issues and a bunch of nodejs projects.

Gauss-Monitoring, 2018-04-12:
Ok thanks I will try npm 5.4.2

altishchenko, 2018-04-17:
Can we close the issue? (@gotthardp, @Gauss-Monitoring  )

closed 2018-04-29 by gotthardp