gotthardp/lorawan-server #366: change version NS to 0.5.3 issue with erlang-base

Gauss-Monitoring, 2018-04-12:
**I have an new issue with the changing version of the NS to 0.5.3 :** 

Unpacking lorawan-server (0.5.3) over (0.5.3) ...
dpkg: dependency problems prevent configuration of lorawan-server:
 lorawan-server depends on erlang-base (>= 1:19.2.1); however:
  Version of erlang-base on system is 1:19.2-1.

dpkg: error processing package lorawan-server (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 lorawan-server

altishchenko, 2018-04-12:
You need to go to the https://www.erlang-solutions.com and follow the instructions on installing their repository on your system. After you are done, you will be able to install OTP 20. Don't forget to apt-get update and apt-get upgrade.

Gauss-Monitoring, 2018-04-12:
alway have the same trouble : 
Depends: erlang-base (= 1:19.2-1) but 1:20.1-1 is to be installed

I really don't understand the dependencies .... How to uninstall all the erlang and install again properly ?

altishchenko, 2018-04-12:
Debian? Ubuntu?
Given that you installed erlang-solutions repository on your system, you can try:
```
sudo -s
dpkg -r erlang
apt-get update # this will update package lists and put ESL erlang first
apt-get upgrade # never hurts
apt-get install erlang
```

Gauss-Monitoring, 2018-04-12:
I follow your instruction and get this message :

# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt-get -f install' to correct these.
The following packages have unmet dependencies:
 lorawan-server : Depends: erlang-base (>= 1:19.2.1) but 1:19.2-1 is installed
E: Unmet dependencies. Try using -f.

altishchenko, 2018-04-12:
It looks like the old server is still in the system. You may want to dpkg -r it, or use -f install to force the installation regardless.

altishchenko, 2018-04-12:
Wait! Why are you trying to install the server before you have erlang 20 on your system? Install erlang 20 first. Remove old erlang, install the one from the erlang solutions repository and only then try installing the server.

Gauss-Monitoring, 2018-04-12:
Try to install the erlang :
~# sudo apt-get install esl-erlang
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  erlang-jinterface libodbc1 libtcl8.6 libtk8.6 libxss1 tcl tcl8.6 tk tk8.6
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  lorawan-server
The following NEW packages will be installed:
  esl-erlang
0 upgraded, 1 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 0 B/49.8 MB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 75001 files and directories currently installed.)
Removing lorawan-server (0.5.3) ...
userdel: user 'lorawan' does not exist
dpkg: error processing package lorawan-server (--remove):
 subprocess installed post-removal script returned error exit status 6
Errors were encountered while processing:
 lorawan-server
E: Sub-process /usr/bin/dpkg returned an error code (1)

altishchenko, 2018-04-12:
First things first.
1. Try removing the server with `dpkg --force-all -r lorawan-server`
2. **Do not install esl-erlang** install just **erlang** instead.
If you properly installed ESL repository on your system, step 2 will fetch the right package.

Gauss-Monitoring, 2018-04-13:
root@******************:/home/ogate# **ls**
1G-LWG-COM-868-XXX  1G-LWS-COM-868-XXX  erlang-solutions_1.0_all.deb  ic880a-gateway  lorawan-server  node-red  package-lock.json  tools

lorawan-server is not empty, and when I try to make a rm -r and apply yes, nothing happened. It seem to be blocked by something else.

root@******************:/home/ogate# **dpkg --force-all -r lorawan-server**
(Reading database ... 75001 files and directories currently installed.)
Removing lorawan-server (0.5.3) ...
userdel: user 'lorawan' does not exist
dpkg: error processing package lorawan-server (--remove):
 subprocess installed post-removal script returned error exit status 6
Errors were encountered while processing:
 lorawan-server

altishchenko, 2018-04-13:
Ok, lorawan-server package installation is garbaged. What you can try is tar xzf lorawa-....deb,
Have a look at install script and create a 'lorawan' user as the script does it. Then retry dpkg -r

altishchenko, 2018-04-14:
well, you just:
```
tar xzf lorawan-server_0.5.3_all.deb
tar xzf control.tar.gz
```
There in the script `postinst` you will see the line:
`useradd --home-dir /var/lib/lorawan-server --create-home $USER`
where USER is set to `lorawan`

You can run this line under `sudo` to recreate the `lorawan` user, after this is done you can try to remove the server package again, there should be no more error:
`userdel: user 'lorawan' does not exist`

altishchenko, 2018-04-17:
@Gauss-Monitoring Did you manage to solve the problem?

gotthardp, 2018-04-29:
I fixed the `userdel: user 'lorawan' does not exist` issue. Otherwise I consider this to be fixed. Thanks.

closed 2018-04-29 by gotthardp