gotthardp/lorawan-server #367: Feature Request: Add ADR Mode configuration setting to Activated (Nodes) page

romansoft, 2018-04-12:
I'm having issue with ADR algorithm being sometimes too aggressive in increasing the data rate while there are missing uplinks. Since in my application, I want to prioritize QoS versus the use of spectrum, I'd like to force some nodes (previously defined with OTAA Profile) with significant uplink losses and low SNR/RSSI to use lower data rates. I can do this on "Activated (Nodes)" page, but the change is ephemeral - after some time ADR kick in again and ups the data rate. What I would like is to have the ADR Mode setting on that screen with Maintain, Disabled, and Auto-Adjust options to take control over ADR. Additionally, I'd like to be able to re-set Max Data Rate on that page once I determine experimentally what the "safe" max limit should be for the deployment environment. 
Right now, the only way to control ADR like this is to change settings in Profile and force device re-join. But, node re-join is not desirable as the nodes may be deployed in hard to reach places and in radio-challenging environment.

gotthardp, 2018-04-12:
I don't want to re-introduce node-specific ADR. It was very cumbersome to edit this when the number of devices increase. You can define a different profile some nodes with different ADR config. You don't need to foce rejoin. Changes in the Profile will immediately affect all nodes that belong to this profile.

romansoft, 2018-04-12:
I was thinking making ADR node-specific ADR may be problematic. But, I did not know the profiles changes have immediate effect without rejoin. If this is the case, then it's reasonable to reject this feature request.
Thanks for clarifying.

On a related note, I found ADR to be quite aggressive with going for the max data rate very quickly, despite some packet loss. For my use cases, I would prefer ADR to prioritize QoS (back off a level down if some packet loss is experienced), but also to be able to reach max data rate if signal is strong and if there was no packet loss for an extended period of time. Would it make sense to add a config parameter that adjusts ADR algorithm behavior (eg. QoS level or a number of allowed missed uplinks or similar proxy)? Is there any other way to achieve this?