gotthardp/lorawan-server #368: ADR implementation prevents LoRaWAN spec compliance in US band

romansoft, 2018-04-12:
The LoRaWAN spec (ver 1.0.2) in section 2.2.2 of LoRaWAN Regional Parameters document states:
"28 If using the over-the-air activation procedure, the end-device should broadcast the JoinReq
29 message alternatively on a random 125 kHz channel amongst the 64 channels defined using
30 DR0 and a random 500 kHz channel amongst the 8 channels defined using DR4. The end
31 device should change channel for every transmission.
32 Personalized devices shall have all 72 channels enabled following a reset."

This means the node must open all 72 channels and let the network server program the node dynamically for the subband supported by the gateway and network. LinkAdrReq is used for this. 

However, the way ADR is implemented, it does not kick in until 50 (or whatever set by user) uplink intervals. This is 50 uplinks that are heard by the server which translates to many more uplinks transmitted by a node all over the frequency band. Add to this the packet loss due to environmental effects, and this "sub-band frequency alignment" can take an unreasonable and unpredictable amount of time (days, if not weeks) with the node having update interval, say, once an hour.  

My proposed solution is to have the server respond with LinkAdrReq upon the first uplink following the OTAA join, repeated on following uplinks until LinkADRAns is received. Data Rate and TX Power should be initially set to minimum and maximum (respectively) until more uplink stats are collected to determine the next appropriate ADR setting. Additionally, the above mechanism should be triggered when there is a significant gap in the frame counter to cover personalized devices and other corner cases.

gotthardp, 2018-04-12:
I assume you use the "maintain" mode. All the ADR "Set" parameters defined in the Profile are (should be) set with the first uplink. The idea is that the server sets this, then waits 50 frames and determines where to go next. This aspect of your request should be implemented. Doesn't it work like this?
Assuming this works, what were the additional thoughts you had? Trigger restart upon detecting a gap in uplinks?

romansoft, 2018-04-12:
I was using Auto-Adjust mode as some nodes may be closer to the gateway than others and want them to adapt individually. When the node opened up all available channels and was using this mode, it would not be heard by the server after join.
I did not try with Maintain mode.

romansoft, 2018-04-12:
Yes, I was thinking about triggering LinkAdrReq after detecting sizable gap in uplink. This is to cover personalized nodes (ABP) or the cases where a node is outside of network coverage for some extended time (say during transport) and re-triggers opening up all channels on its end after it detects loss of the gateway.