gotthardp/lorawan-server #370: Web admin interface doesn't work

stefanobimbo, 2018-04-13:
I upgraded from version 0.4.3 to 0.5.3. After that, the web administration interface didn't work anymore. I see all tabs in the right column, but when I click on every tab, the "NOT FOUND" page is showed.

Then I purged lorawan-server packet and installed again from scratch the 0.5.3 version. The same happens.

The debug.log follows:

2018-04-13 16:39:31.889 [debug] <0.118.0>@lager_handler_watcher:119 Lager installed handler error_logger_lager_h into error_logger
2018-04-13 16:39:31.896 [debug] <0.102.0> Supervisor gr_param_sup started gr_param:start_link(gr_lager_default_tracer_params) at pid <0.120.0>
2018-04-13 16:39:31.897 [debug] <0.101.0> Supervisor gr_counter_sup started gr_counter:start_link(gr_lager_default_tracer_counters) at pid <0.121.0>
2018-04-13 16:39:31.897 [debug] <0.103.0> Supervisor gr_manager_sup started gr_manager:start_link(gr_lager_default_tracer_params_mgr, gr_lager_default_tracer_params, []) at pid <0.122.0>
2018-04-13 16:39:31.898 [debug] <0.103.0> Supervisor gr_manager_sup started gr_manager:start_link(gr_lager_default_tracer_counters_mgr, gr_lager_default_tracer_counters, [{input,0},{filter,0},{output,0},{job_input,0},{job_run,0},{job_time,0},{job_error,0}]) at pid <0.123.0>
2018-04-13 16:39:32.088 [info] <0.31.0> Application lager started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.095 [info] <0.31.0> Application crypto started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.096 [info] <0.31.0> Application cowlib started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.115 [info] <0.31.0> Application asn1 started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.115 [info] <0.31.0> Application public_key started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.132 [debug] <0.132.0> Supervisor ssl_sup started ssl_manager:start_link([]) at pid <0.133.0>
2018-04-13 16:39:32.137 [debug] <0.132.0> Supervisor ssl_sup started tls_connection_sup:start_link() at pid <0.134.0>
2018-04-13 16:39:32.138 [debug] <0.132.0> Supervisor ssl_sup started ssl_listen_tracker_sup:start_link() at pid <0.135.0>
2018-04-13 16:39:32.140 [debug] <0.132.0> Supervisor ssl_sup started dtls_connection_sup:start_link() at pid <0.136.0>
2018-04-13 16:39:32.141 [debug] <0.132.0> Supervisor ssl_sup started dtls_udp_sup:start_link() at pid <0.137.0>
2018-04-13 16:39:32.141 [info] <0.31.0> Application ssl started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.143 [debug] <0.141.0> Supervisor ranch_sup started ranch_server:start_link() at pid <0.142.0>
2018-04-13 16:39:32.143 [info] <0.31.0> Application ranch started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.145 [debug] <0.146.0> Supervisor cowboy_sup started cowboy_clock:start_link() at pid <0.147.0>
2018-04-13 16:39:32.146 [info] <0.31.0> Application cowboy started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.148 [info] <0.31.0> Application gun started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.149 [info] <0.31.0> Application jsx started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.150 [info] <0.31.0> Application iso8601 started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.153 [info] <0.31.0> Application xmerl started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.154 [info] <0.31.0> Application recon started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.154 [info] <0.31.0> Application rabbit_common started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.158 [info] <0.31.0> Application amqp_client started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.160 [info] <0.31.0> Application emqttc started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.164 [debug] <0.165.0> Supervisor mongodb_supervisor started mongodb:start() at pid <0.166.0>
2018-04-13 16:39:32.166 [info] <0.31.0> Application erlmongo started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.171 [info] <0.31.0> Application mnesia exited with reason: stopped
2018-04-13 16:39:32.171 [info] <0.169.0>@lorawan_db:ensure_tables:20 Database create schema
2018-04-13 16:39:32.189 [debug] <0.56.0> Supervisor kernel_safe_sup started disk_log_sup:start_link() at pid <0.170.0>
2018-04-13 16:39:32.189 [debug] <0.56.0> Supervisor kernel_safe_sup started disk_log_server:start_link() at pid <0.171.0>
2018-04-13 16:39:32.202 [debug] <0.183.0> Supervisor mnesia_sup started mnesia_sup:start_event() at pid <0.184.0>
2018-04-13 16:39:32.202 [debug] <0.183.0> Supervisor mnesia_sup started mnesia_ext_sup:start() at pid <0.185.0>
2018-04-13 16:39:32.202 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_monitor:start() at pid <0.187.0>
2018-04-13 16:39:32.202 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_subscr:start() at pid <0.188.0>
2018-04-13 16:39:32.202 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_locker:start() at pid <0.189.0>
2018-04-13 16:39:32.202 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_recover:start() at pid <0.190.0>
2018-04-13 16:39:32.211 [debug] <0.56.0> Supervisor kernel_safe_sup started dets_sup:start_link() at pid <0.198.0>
2018-04-13 16:39:32.212 [debug] <0.56.0> Supervisor kernel_safe_sup started dets_server:start_link() at pid <0.199.0>
2018-04-13 16:39:32.222 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_tm:start() at pid <0.191.0>
2018-04-13 16:39:32.222 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_checkpoint_sup:start() at pid <0.211.0>
2018-04-13 16:39:32.222 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_snmp_sup:start() at pid <0.212.0>
2018-04-13 16:39:32.222 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_controller:start() at pid <0.213.0>
2018-04-13 16:39:32.223 [debug] <0.186.0> Supervisor mnesia_kernel_sup started mnesia_late_loader:start() at pid <0.214.0>
2018-04-13 16:39:32.223 [debug] <0.183.0> Supervisor mnesia_sup started mnesia_kernel_sup:start() at pid <0.186.0>
2018-04-13 16:39:32.223 [info] <0.169.0>@lorawan_db:create_table:126 Database create users
2018-04-13 16:39:32.223 [info] <0.31.0> Application mnesia started on node 'lorawan@ubuntu-dev'
2018-04-13 16:39:32.238 [info] <0.169.0>@lorawan_db:set_defaults:226 Database create default user:password
2018-04-13 16:39:32.250 [info] <0.169.0>@lorawan_db:create_table:126 Database create servers
2018-04-13 16:39:32.258 [info] <0.169.0>@lorawan_db:create_table:126 Database create networks
2018-04-13 16:39:32.268 [info] <0.169.0>@lorawan_db:create_table:126 Database create gateways
2018-04-13 16:39:32.278 [info] <0.169.0>@lorawan_db:create_table:126 Database create multicast_channels
2018-04-13 16:39:32.288 [info] <0.169.0>@lorawan_db:create_table:126 Database create profiles
2018-04-13 16:39:32.299 [info] <0.169.0>@lorawan_db:create_table:126 Database create devices
2018-04-13 16:39:32.309 [info] <0.169.0>@lorawan_db:create_table:126 Database create nodes
2018-04-13 16:39:32.319 [info] <0.169.0>@lorawan_db:create_table:126 Database create ignored_nodes
2018-04-13 16:39:32.329 [info] <0.169.0>@lorawan_db:create_table:126 Database create pending
2018-04-13 16:39:32.339 [info] <0.169.0>@lorawan_db:create_table:126 Database create txframes
2018-04-13 16:39:32.349 [info] <0.169.0>@lorawan_db:create_table:126 Database create rxframes
2018-04-13 16:39:32.359 [info] <0.169.0>@lorawan_db:create_table:126 Database create connectors
2018-04-13 16:39:32.369 [debug] <0.112.0>@lager_handler_watcher:119 Lager installed handler lager_backend_throttle into lager_event
2018-04-13 16:39:32.370 [info] <0.169.0>@lorawan_db:create_table:126 Database create handlers
2018-04-13 16:39:32.380 [info] <0.169.0>@lorawan_db:create_table:126 Database create events
2018-04-13 16:39:32.396 [debug] <0.322.0> Supervisor {<0.322.0>,ranch_listener_sup} started ranch_conns_sup:start_link(http, supervisor, 5000, ranch_tcp, 5000, cowboy_clear) at pid <0.323.0>
2018-04-13 16:39:32.397 [debug] <0.324.0> Supervisor {<0.324.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.6592>, ranch_tcp, <0.323.0>) at pid <0.325.0>
2018-04-13 16:39:32.398 [debug] <0.324.0> Supervisor {<0.324.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.6592>, ranch_tcp, <0.323.0>) at pid <0.326.0>
2018-04-13 16:39:32.398 [debug] <0.324.0> Supervisor {<0.324.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.6592>, ranch_tcp, <0.323.0>) at pid <0.327.0>
2018-04-13 16:39:32.398 [debug] <0.324.0> Supervisor {<0.324.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.6592>, ranch_tcp, <0.323.0>) at pid <0.328.0>
[... cut ...]
2018-04-13 16:39:32.406 [debug] <0.335.0> Supervisor {<0.335.0>,lorawan_sup} started lorawan_gw_sup:start_link() at pid <0.337.0>
2018-04-13 16:39:32.406 [debug] <0.335.0> Supervisor {<0.335.0>,lorawan_sup} started lorawan_http_registry:start_link() at pid <0.341.0>
2018-04-13 16:39:32.408 [debug] <0.341.0>@lorawan_http_registry:compile_routes:71 New routes []
2018-04-13 16:39:32.409 [debug] <0.342.0> Supervisor {<0.342.0>,lorawan_backend_sup} started lorawan_connector_sup:start_link() at pid <0.343.0>

stefanobimbo, 2018-04-13:
Well, tabs are in the column on the left of the web page, not on the right...

gotthardp, 2018-04-13:
Refresh you web browser cache.

stefanobimbo, 2018-04-16:
Now it's OK.
thanks

closed 2018-04-16 by gotthardp