gotthardp/lorawan-server #371: Help for erlang fun

webfrank, 2018-04-14:
Hi,
   I have a node which sends binary payload with conditional information:

```
                     CID          VAL
frame 1: (HEX) 11 0A 0402 0000 29 0874
frame 2: (HEX) 11 0A 0405 0000 21 1568
```
for frame 1 CID==0402, VAL (0874) is signed int16
for frame 2 CID==0405, VAL (1568) is unsigned int16

In decoding fun I would like to parse up to CID into variables and put the rest on a new binary variable which I could parse later based on CID value. After second parsing I would like to return a JSON with field "temp" for frame 1 and "rh" for frame 2.

I think I should use "case" statement but I'm not familiar with erlang and I'm looking for your help.

altishchenko, 2018-04-14:
Hello @webfrank I am sure you don't need case statement, as for this type of things Erlang uses 'pattern matching' in function closes. Before I can help you, I'd like to know what is the meaning of the `0000 29` or `0000 21` in your output. Can it be safely ignored? Or matched precisely?

altishchenko, 2018-04-14:
To give you the feel:
```erlang
fun (Fields, <<17, 10, 16#0402:16, _Ignore:18, Val:16/signed>>) ->
          Fields#{temp = Val}; % return temp if CID is 0402
    (Fields, <<17, 10, 16#0405:16, _Ignore:18, Val:16>>) ->
          Fields#{rh = Val} % return rh if CIS is 0405
end.
```

webfrank, 2018-04-14:
Hi, it could be ignored 

Inviato da iPhone

gotthardp, 2018-04-19:
Documented. Thank you both!

closed 2018-04-19 by gotthardp