gotthardp/lorawan-server #372: Feature request: Gateway Timestamp

webfrank, 2018-04-16:
From Semtec Packet Forwader protocol the network server will receive three timestamps:

'''
 time | string | UTC time of pkt RX, us precision, ISO 8601 'compact' format
 tmms | number | GPS time of pkt RX, number of milliseconds since 06.Jan.1980
 tmst | number | Internal timestamp of "RX finished" event (32b unsigned)
'''

but on mqtt payload I have only the first. I would like to receive also tmms (if available) and tmst which is always available.

gotthardp, 2018-04-16:
I wonder why? The tmst is an internal gateway number used only to schedule downlinks.

webfrank, 2018-04-17:
Hi, I agrre with you but I was looking for a timestamp useful to 
geolocate a node. Is it possible to have a field with epoch with micro 
seconds as an mqtt field?

I would like to triangulate using RSSI and timestamp of received packet.

gotthardp, 2018-04-29:
I added the `tmms` field. Would you be able to test it before I do a release?

webfrank, 2018-04-29:
Yes, I’ll test soon

Inviato da iPhone

gotthardp, 2018-05-16:
in 0.5.4

closed 2018-05-16 by gotthardp

webfrank, 2018-05-17:
In 0.5.5 Docker is not present

gotthardp, 2018-05-17:
I see both the `release-0.5.4` and the `latest` tag.