gotthardp/lorawan-server #375: Login server on mobile

Apaisal, 2018-04-19:
[image]
Login server on mobile and login popup page still popup.
I tested this issue on samsung galaxy A8 with google chrome app and same result on samsung browser.

gotthardp, 2018-04-19:
True,

gotthardp, 2018-04-29:
I made a fix that could help. Would you be able to test the latest master, or you need to wait for a release?

Apaisal, 2018-04-30:
Thank you. I will test it. What is commit id to valid?

gotthardp, 2018-04-30:
https://github.com/gotthardp/lorawan-server/commit/99dcb729d3e99a79905918c63dde8f1666867a0f

gotthardp, 2018-06-26:
I did not forget about this. I just don't know where is the problem. I keep trying.

gotthardp, 2018-08-10:
@Apaisal, I may have fixed this problem. It would be great if you could pull the latest master and check it out. (I recommend starting it from a new directory so you don't destroy your existing database,)

gotthardp, 2018-08-16:
I tested this on many devices and I believe it is really fixed now.

closed 2018-08-16 by gotthardp