gotthardp/lorawan-server #378: no mqtt publish upon u/l receive

stefanobimbo, 2018-04-23:
I configured a connector to publish received u/l messages in an mqtt topic (I use mosquitto), but my subscribers don't receive anything when an u/l message is received (I am sure the message is received because I see it in the Received Frames).
Using moquitto_pub I tried to publish something manually to the same topic and I am sure the mqtt broker is working OK (every subscribed client received it).
I am a bit confused about connector authentication. should I use Username+Password with related fields filled up with something that is not an empty string? Should I configure mosquitto to use the same user/password in its config file?

Is there something I can give you to check my lorawan-server configuration parameters?

thanks

gotthardp, 2018-04-23:
Any errors in the server log? By default no username/password is required.
You can share with me either the screenshots, or the JSON files http://server/api/handlers and http://server/api/connectors

stefanobimbo, 2018-04-24:
Nothing in the server log. Looking at the mosquitto log, I don't see connections from server, so I think is can be a configuration problem.
I use a device that joins OTAA to network, but the same problem is for ABP nodes. All configured parameters follows:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

stefanobimbo, 2018-04-24:
Removing failed flags works OK.
There's only one thing to remark: why could I set flags from web interface?

stefanobimbo, 2018-04-25:
yes, issue is closed.

many thanks

closed 2018-04-25 by stefanobimbo