gotthardp/lorawan-server #381: node-gateway communication

gmbankar37, 2018-04-30:
how to make my device communicate with only one particular gateway (If there are more than one on the server) ? Do I need to define different network with different NetID ?

gotthardp, 2018-04-30:
What do you want to achieve? The gateways are just like an antenna. The server selects the gateway with the best signal. Why would you need to use another one with worse signal?

gmbankar37, 2018-05-02:
No I need a dedicated gateway for a number of nodes. I want to send uplink signal through a specific gateway only, creating a private network.

Apaisal, 2018-05-02:
Hello @gmbankar37

Please note!!!

Private network is able to setup sync word by gateway and end device.

Network server just routing and handle your data information to your
application via your network id and connector.

cristianhumelnicu, 2018-05-03:
@gmbankar37  The stream from lorawan node will be received by all gateway from area . If exist gateways owned by other people, or owned by you ,  them will receive that stream too .. but they will be unable to decode the messages .
The best approach is to connect all your gateways to same lorawan server , and let the server to do packet deduplication 

Is not matter if stream data from node will be received by all your gateways because authentification is done at  lorawan application server

cristianhumelnicu, 2018-05-03:
@gmbankar37 If you still wanna have separate networks try to install more than one lorawan server , one form each gateways ...

gotthardp, 2018-05-03:
That was also my thinking. If it is OK to share the server (operate more networks by a single server), it shall be also OK to share the gateways. If it's not OK to share the gateways among several networks, one can easily create a new server instance and have everything isolated.
I didn't find a reason why anyone would share the server, but not the gateways.

gmbankar37, 2018-05-04:
> Petr ... Is hard to operate same gateway with two or more servers ... what
> will you do if node have OTAA authentification ?
>
> I really do not see any problem if stream from lora nodes will be recieved
> by more than one gateway as long them are operate by same lorawan server ...
> Is a false problem

@cristianhumelnicu yes I also arrived at the same solution to have two
different servers after thinking about it. Also the main purpose of LoRaWAN
is not fulfilled if we have to use single gateway even if there are more
than one in the network. Because its the gateways that make the network.
Thank you for all your answers.

cristianhumelnicu, 2018-05-04:
@gmbankar37 BTW what distance is between that two gateways ? And ..what model of gateways is ?

gmbankar37, 2018-05-04:
I am not yet sure of the distance between the two gateways. The gateway
model I am using is RAK 831 from Rakwireless.

cristianhumelnicu, 2018-05-04:
@gmbankar37 Try first time with one single  lorawanserver and two gateways connected at same server. 

And if you find something that do not like then try to install one server on each gateway .
Like if told you before .. there is no reason to to concern if data stream from lorawan node will be received by two gateway managed by same lora-wan server. The server will be able to do data deduplication after athentification, even if the data will be forwarded by two or more gateway in same time ...

Hint: Just thing about it how TTN Network manage a lot of gateways with same server ...

gotthardp, 2018-05-07:
To avoid RF "interference" by two networks, you will have to operate each of them on a completely different set of channels/frequencies. The standard allows that for US 920, but not for EU 868 as the first 3 channels in EU868 are "mandatory". You could e.g. setup one network at 868 and another on 433.