gotthardp/lorawan-server #382: How to clean up old data ( frames)

pdl86, 2018-05-03:
Hello,

I have noticed that a gateway with Lorawan server embeded on it ( in that case a Multitech Conduit gateway) , has its web interface becoming very slow  and even part of it unusable : Main dashboard, frame page,  when the gateway is up and running since week and months ( here 3 months). It seems that the received frames are never clean up from the database, even if processed by an handler and a connector.

Is there a way to clean up selectively such past frames ( aka all frames older than xx days, for all or a given devices, ... ) ? via API call ? ( if so it can be schedule) 

Thanks in advance .
Philippe

PS Petr , as I'm still having the "segmentation  fault" issue, I can not export / import the db,  so my only option so far is to rebuild the gateway ( delete the database and re-install for recreating it)

gotthardp, 2018-05-03:
The serve keeps last X frames for each node. Old frames shall be deleted automatically. Try decreasing the `retained_rxframes` config variable.

pdl86, 2018-05-04:
Hello,
as a work around, I've move the rxframes and events, DCD and DCL files in the Mnesia folder ( to .old ) and restart teh lorawan-server, and I have access again  to the dashbaord and rxfrances pages.

the file had the following size : 
-rw-r--r--    1 root     root       4308047 May  3 15:59 rxframes.DCD.old
-rw-r--r--    1 root     root         58148 May  4 08:50 events.DCD.old
-rw-r--r--    1 root     root         10948 May  4 11:48 events.DCL.old
-rw-r--r--    1 root     root        723510 May  4 11:51 rxframes.DCL.old

but this is not really "nice" .

regards
Philippe

gotthardp, 2018-05-04:
Well, you can always set the `retained_rxframes` to a lower value.

pdl86, 2018-05-04:
Hello 

effectively the limit does not seem to be  take into account : 
[image]

after 54 minutes, 54 received frames "visible" .

regards
Philippe

gotthardp, 2018-05-04:
The tables are trimmed every 60 minutes, so it may take upto 2 hours until the frames are trimmed.

pdl86, 2018-05-04:
Hi
Ok , seems to be the case.
So I will follow up to see what is "piling" up to make the db increasing
so much.

at the first glance , a bunch of nodes , in the vincinity

regards
philippe

gotthardp, 2018-05-05:
Could it be due to nodes from some other networks? Frames from unknown devices are logged too. I could disable this feature to even more reduce the database size.

pdl86, 2018-05-05:
hi ,

you are right: after one day almost 1000 frames most of them independent.
not logging unknown  frames as an option would by a very good idea. ( nice
to be able to (re)activate it for troubleshooting during some time).

would be nice too to be able to select frames a) for "massively"  mark such
nodes as to be ignored
b) delete such frame

philipe

gotthardp, 2018-05-05:
You can always put nodes into the "ignored nodes". Such nodes are not logged. Could this solve a)?

pdl86, 2018-05-05:
yes but selecting manually  hundreds of nodes is boring  ! so selecting at
once a bunch of them will save a lot of time

regards
philippe

pdl86, 2018-05-07:
> As a starting point, I implemented real-time server-wide config
> parameter to enable/disable logging of ignored frames. It will be in
> 0.5.4 in the Admin Menu under Servers as "Log Ignored?"

Hi
many thanks . this it is  a "real" good starting point.
regards
philppe

pdl86, 2018-05-11:
Hello 

associated  to you improvment : "Log ignored", it would be useful to automatically assigned  not commissioned / not activted  devices as "Ignored" (set-up globally) , or via a page select ( as it was to be done in #248 on "events" , but also on the "received frames" page ).
thanks in advance
Philippe

gotthardp, 2018-08-25:
I believe the 0.6.0 solved this issue.

closed 2018-08-25 by gotthardp