gotthardp/lorawan-server #383: gateway is deleted from NS is still accepting uplinks from nodes

gotthardp, 2018-05-04:
@romansoft, let's create a new issue for every new issue.
I found another issue: when a gateway is deleted from NS, it is still accepting uplinks from nodes. This is unexpected and undesired behavior.

gotthardp, 2018-05-04:
What do you mean by "it is still accepting uplinks"?

romansoft, 2018-05-04:
During work on issue https://github.com/gotthardp/lorawan-server/issues/355, I deleted the gateway that was supporting imcompatible pcaket forwarder and corrupting the state of nodes in NS. After cleaning up the node state in NS (by re-joining the nodes), when I later reconnected the gateway to the internet, NS was still accepting uplinks from nodes through this gateway and corrupting/crashing the server as before. You can see the offending gateway showing up on the Activated (Nodes) detail listing:
[image]

gotthardp, 2018-05-04:
Yeah, this feature used to be there in 0.4 and somehow disappeared in 0.5.
I put it back again, so the server shall ignore frames from unknown gateways. It's also a security feature.