gotthardp/lorawan-server #385: Australian region configuration issues

romansoft, 2018-05-08:
I set up the network for Australia region and there are issues with it. 

Firstly, the Data Rate settings are mismatched. What I mean is when SF/BW settings are defined in ADR, they result in wrong DR commands sent to a node. For example, in debug log, I see that DR 0 sent to device corresponds to SF12/125kHz, instead of SF10/125kHz, and so on for other DR settings. That is DR definitions are shifted with respect to SF definitions. 

2018-05-07 16:43:11.961 [debug] <0.23685.1>@lorawan_mac_commands:send_adr:332 LinkADRReq {5,3,[{8,15},{65,65}]}
2018-05-07 16:43:11.961 [debug] <0.23685.1>@lorawan_mac_commands:build_fopts:74 03D994F9 <- [{link_adr_req,3,5,2,7,0},{link_adr_req,3,5,65280,0,0}]
2018-05-07 16:43:56.393 [debug] <0.23729.1>@lorawan_mac_commands:handle_fopts:16 03D994F9 -> [{link_adr_ans,1,1,1},{link_adr_ans,1,1,1}]
2018-05-07 16:43:56.393 [debug] <0.23729.1>@lorawan_mac_commands:handle_adr:159 LinkADRReq 03D994F9 succeeded
2018-05-07 16:43:56.393 [debug] <0.23729.1>@lorawan_mac_commands:build_fopts:74 03D994F9 <- [dev_status_req]
2018-05-07 16:47:40.166 [debug] <0.23898.1>@lorawan_mac_commands:handle_fopts:16 03D994F9 -> [{dev_status_ans,165,6}]
2018-05-07 16:47:40.166 [debug] <0.23898.1>@lorawan_mac_commands:store_actual_adr:139 DataRate 03D994F9 switched to dr 5
2018-05-07 16:47:40.166 [debug] <0.23898.1>@lorawan_mac_commands:handle_status:219 DevStatus: battery 165, margin: 6 (max -7.5)
2018-05-07 16:47:59.331 [warning] <0.22912.1> server {http_error,{412,"/api/nodes/03D994F9"}}
2018-05-07 16:48:28.795 [debug] <0.23968.1>@lorawan_mac_commands:send_adr:332 LinkADRReq {5,2,[{8,15},{65,65}]}
2018-05-07 16:48:28.795 [debug] <0.23968.1>@lorawan_mac_commands:build_fopts:74 03D994F9 <- [{link_adr_req,2,5,2,7,0},{link_adr_req,2,5,65280,0,0}]
2018-05-07 16:50:16.830 [warning] <0.24038.1> node 039EB91B {uplinks_missed,1}
2018-05-07 16:50:27.310 [debug] <0.24050.1>@lorawan_mac_commands:handle_fopts:16 03D994F9 -> [{link_adr_ans,1,1,1},{link_adr_ans,1,1,1}]
2018-05-07 16:50:27.310 [debug] <0.24050.1>@lorawan_mac_commands:store_actual_adr:139 DataRate 03D994F9 switched to dr 4
2018-05-07 16:50:27.310 [debug] <0.24050.1>@lorawan_mac_commands:handle_adr:159 LinkADRReq 03D994F9 succeeded
2018-05-07 16:50:27.310 [debug] <0.24050.1>@lorawan_mac_commands:build_fopts:74 03D994F9 <- [dev_status_req]
2018-05-07 16:52:07.869 [debug] <0.24150.1>@lorawan_mac_commands:handle_fopts:16 03D994F9 -> [{dev_status_ans,166,8}]
2018-05-07 16:52:07.869 [debug] <0.24150.1>@lorawan_mac_commands:store_actual_adr:139 DataRate 03D994F9 switched to dr 4
2018-05-07 16:52:07.869 [debug] <0.24150.1>@lorawan_mac_commands:handle_status:219 DevStatus: battery 166, margin: 8 (max -10.0)
2018-05-07 16:52:34.878 [warning] <0.22912.1> server {http_error,{412,"/api/nodes/03D994F9"}}
2018-05-07 16:53:03.143 [debug] <0.24209.1>@lorawan_mac_commands:send_adr:332 LinkADRReq {5,0,[{8,15},{65,65}]}
2018-05-07 16:53:03.143 [debug] <0.24209.1>@lorawan_mac_commands:build_fopts:74 03D994F9 <- [{link_adr_req,0,5,2,7,0},{link_adr_req,0,5,65280,0,0}]
2018-05-07 16:53:20.792 [debug] <0.24222.1>@lorawan_mac_commands:handle_fopts:16 03D994F9 -> [{link_adr_ans,1,1,1},{link_adr_ans,1,1,1}]
2018-05-07 16:53:20.792 [debug] <0.24222.1>@lorawan_mac_commands:store_actual_adr:139 DataRate 03D994F9 switched to dr 2
2018-05-07 16:53:20.792 [debug] <0.24222.1>@lorawan_mac_commands:handle_adr:159 LinkADRReq 03D994F9 succeeded
2018-05-07 16:53:20.792 [debug] <0.24222.1>@lorawan_mac_commands:build_fopts:74 03D994F9 <- [dev_status_req]
2018-05-07 16:53:24.375 [debug] <0.24225.1>@lorawan_mac_commands:handle_fopts:16 03D994F9 -> [{dev_status_ans,165,6}]
2018-05-07 16:53:24.375 [debug] <0.24225.1>@lorawan_mac_commands:store_actual_adr:139 DataRate 03D994F9 switched to dr 2
2018-05-07 16:53:24.375 [debug] <0.24225.1>@lorawan_mac_commands:handle_status:219 DevStatus: battery 165, margin: 6 (max -15.0)
 
Secondly, adding the new AU region is now creating errors for US network nodes. Here is the debug log excerpt for one of US nodes. This happens on missed node uplinks. After restarting the NS, the issue went away, but I suspect it will start again after connecting AU gateway again.

debug.log :
2018-05-08 18:04:22.299 [warning] <0.12948.3> node 03BE93C3 {uplinks_missed,1}
2018-05-08 18:04:22.500 [debug] <0.12948.3>@lorawan_mac_commands:build_fopts:74 03BE93C3 <- [dev_status_req]
2018-05-08 18:04:22.500 [error] <0.12948.3>@lorawan_mac_region:f2ch:55 gen_statem <0.12948.3> in state uplink terminated with reason: no function clause matching lorawan_mac_region:f2ch(903.3, {9152,2}, {9159,16}) line 55
2018-05-08 18:04:22.501 [error] <0.12948.3>@lorawan_mac_region:f2ch:55 CRASH REPORT Process <0.12948.3> with 0 neighbours crashed with reason: no function clause matching lorawan_mac_region:f2ch(903.3, {9152,2}, {9159,16}) line 55
2018-05-08 18:04:22.501 [error] <0.12804.3>@lorawan_mac_region:f2ch:55 Supervisor lorawan_handler_sup had child handler started with lorawan_handler:start_link() at <0.12948.3> exit with reason no function clause matching lorawan_mac_region:f2ch(903.3, {9152,2}, {9159,16}) line 55 in context child_terminated

gotthardp, 2018-05-08:
Could you please provide more details on what is shifted where?

The second error is caused because a device configured for the australian region sends data on an unexpected frequency.

romansoft, 2018-05-08:
Currently, the AU gateway and devices are disconnected, but the errors still appear for US network devices (which did not change their frequency plan).

gotthardp, 2018-05-08:
If you correctly switched to US it cannot be the same errors. You either didn't switch or it's not the same errors.

romansoft, 2018-05-08:
I see now why US devices report this error. Somehow, they show wrong channels being used (the channels assigned for AU network) despite using US profile. The AU and US profiles should be independent from each other, but, for some reason, AU profile unexplainably "affected" the US devices. I'll try to manually set the channels to the correct ones in ADR setting, and see if it fixes the errors.
 
[image]

romansoft, 2018-05-08:
Strangely, I also see US devices that have correct channel configuration creating errors. Here is en example:

2018-05-08 00:10:35.907 [debug] <0.12231.2>@lorawan_mac_commands:build_fopts:74 03206C11 <- [dev_status_req]
2018-05-08 00:10:35.908 [error] <0.12231.2>@lorawan_mac_region:f2ch:55 gen_statem <0.12231.2> in state uplink terminated with reason: no function clause matching lorawan_mac_region:f2ch(902.3, {9152,2}, {9159,16}) line 55
2018-05-08 00:10:35.908 [error] <0.12231.2>@lorawan_mac_region:f2ch:55 CRASH REPORT Process <0.12231.2> with 0 neighbours crashed with reason: no function clause matching lorawan_mac_region:f2ch(902.3, {9152,2}, {9159,16}) line 55
2018-05-08 00:10:35.908 [error] <0.11576.2>@lorawan_mac_region:f2ch:55 Supervisor lorawan_handler_sup had child handler started with lorawan_handler:start_link() at <0.12231.2> exit with reason no function clause matching lorawan_mac_region:f2ch(902.3, {9152,2}, {9159,16}) line 55 in context child_terminated
2018-05-08 00:10:39.624 [debug] <0.12236.2>@lorawan_mac_commands:build_fopts:74 03206C11 <- [dev_status_req]
2018-05-08 00:10:39.625 [error] <0.12236.2>@lorawan_mac_region:f2ch:55 gen_statem <0.12236.2> in state uplink terminated with reason: no function clause matching lorawan_mac_region:f2ch(903.5, {9152,2}, {9159,16}) line 55
2018-05-08 00:10:39.625 [error] <0.12236.2>@lorawan_mac_region:f2ch:55 CRASH REPORT Process <0.12236.2> with 0 neighbours crashed with reason: no function clause matching lorawan_mac_region:f2ch(903.5, {9152,2}, {9159,16}) line 55
2018-05-08 00:10:39.625 [error] <0.11576.2>@lorawan_mac_region:f2ch:55 Supervisor lorawan_handler_sup had child handler started with lorawan_handler:start_link() at <0.12236.2> exit with reason no function clause matching lorawan_mac_region:f2ch(903.5, {9152,2}, {9159,16}) line 55 in context child_terminated

[image]

The AU network configuration and device profile were copied from corresponding US ones and modified accordingly. Is it possible this corrupted something?

romansoft, 2018-05-08:
OK, I found the cause of these errors. The US device profile had AU network config setting in it. I'm not sure why it happened, but probably it was my mistake. I'll let it run longer to see if the errors are cleared now. 

As to the first error, I don't have the AU gateway and devices anymore, so I'll describe from my memory. When I defined device profile ADR settings to SF10/125KHz, NS was sending command to device with DR4 setting. I had to change ADR setting to SF12/125kHz to make NS send DR 0 setting. (SF12/125kHz should not be even showing up as a possible option to choose for AU region.)
This is incorrect according to LoRaWAN regionals spec:

[image]

gotthardp, 2018-05-08:
Thanks. I will double check the algorithm.

gotthardp, 2018-05-18:
Well, you have some old DR table. The table in 1.0.2 as well as in the latest 1.1b is the following:
[image]

romansoft, 2018-05-19:
You are correct. Indeed, the later update to 1.0.2 version changed the definition of DR, despide both revisions of spec being called "Final"... It was the LoRaWAN spec issue.

closed 2018-05-19 by romansoft