gotthardp/lorawan-server #386: HTTPS not working right with Chrome and Firefox

romansoft, 2018-05-09:
I recently started having problems with using HTTPS to connect with the dashboard on Chrome and Firefox. Used to work just fine till a week or so ago. MS Edge still works OK. The SSL certificates (from letsencrypt) are valid.
Chrome shows this error:

[image]

 Firefox is a more descriptive:

[image]

Here is the debug.log for starting up the server:

2018-05-09 04:43:00.542 [debug] <0.203.0>@lager_handler_watcher:119 Lager installed handler {lager_file_backend,"error.log"} into lager_event
2018-05-09 04:43:00.548 [debug] <0.205.0>@lager_handler_watcher:119 Lager installed handler error_logger_lager_h into error_logger
2018-05-09 04:43:00.559 [debug] <0.176.0> Supervisor gr_param_sup started gr_param:start_link(gr_lager_default_tracer_params) at pid <0.207.0>
2018-05-09 04:43:00.559 [debug] <0.171.0> Supervisor gr_counter_sup started gr_counter:start_link(gr_lager_default_tracer_counters) at pid <0.208.0>
2018-05-09 04:43:00.560 [debug] <0.179.0> Supervisor gr_manager_sup started gr_manager:start_link(gr_lager_default_tracer_params_mgr, gr_lager_default_tracer_params, []) at pid <0.209.0>
2018-05-09 04:43:00.560 [debug] <0.179.0> Supervisor gr_manager_sup started gr_manager:start_link(gr_lager_default_tracer_counters_mgr, gr_lager_default_tracer_counters, [{input,0},{filter,0},{output,0},{job_input,0},{job_run,0},{job_time,0},{job_error,0}]) at pid <0.210.0>
2018-05-09 04:43:00.721 [info] <0.33.0> Application lager started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.725 [info] <0.33.0> Application crypto started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.726 [info] <0.33.0> Application cowlib started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.732 [info] <0.33.0> Application asn1 started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.733 [info] <0.33.0> Application public_key started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.743 [debug] <0.220.0> Supervisor ssl_admin_sup started ssl_pem_cache:start_link([]) at pid <0.221.0>
2018-05-09 04:43:00.749 [debug] <0.220.0> Supervisor ssl_admin_sup started ssl_manager:start_link([]) at pid <0.222.0>
2018-05-09 04:43:00.749 [debug] <0.219.0> Supervisor ssl_sup started ssl_admin_sup:start_link() at pid <0.220.0>
2018-05-09 04:43:00.753 [debug] <0.223.0> Supervisor ssl_connection_sup started tls_connection_sup:start_link() at pid <0.224.0>
2018-05-09 04:43:00.754 [debug] <0.223.0> Supervisor ssl_connection_sup started ssl_listen_tracker_sup:start_link() at pid <0.225.0>
2018-05-09 04:43:00.760 [debug] <0.223.0> Supervisor ssl_connection_sup started dtls_connection_sup:start_link() at pid <0.226.0>
2018-05-09 04:43:00.761 [debug] <0.223.0> Supervisor ssl_connection_sup started dtls_udp_sup:start_link() at pid <0.227.0>
2018-05-09 04:43:00.761 [debug] <0.219.0> Supervisor ssl_sup started ssl_connection_sup:start_link() at pid <0.223.0>
2018-05-09 04:43:00.761 [info] <0.33.0> Application ssl started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.763 [debug] <0.231.0> Supervisor ranch_sup started ranch_server:start_link() at pid <0.232.0>
2018-05-09 04:43:00.763 [info] <0.33.0> Application ranch started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.765 [debug] <0.236.0> Supervisor cowboy_sup started cowboy_clock:start_link() at pid <0.237.0>
2018-05-09 04:43:00.766 [info] <0.33.0> Application cowboy started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.772 [info] <0.33.0> Application gun started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.773 [info] <0.33.0> Application jsx started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.773 [info] <0.33.0> Application iso8601 started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.776 [info] <0.33.0> Application xmerl started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.777 [info] <0.33.0> Application recon started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.777 [info] <0.33.0> Application rabbit_common started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.786 [info] <0.33.0> Application amqp_client started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.787 [info] <0.33.0> Application emqttc started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.794 [debug] <0.255.0> Supervisor mongodb_supervisor started mongodb:start() at pid <0.256.0>
2018-05-09 04:43:00.800 [info] <0.33.0> Application erlmongo started on node 'lorawan@ip-172-31-27-175'
2018-05-09 04:43:00.807 [debug] <0.275.0> Supervisor {<0.275.0>,ranch_listener_sup} started ranch_conns_sup:start_link(http, supervisor, 5000, ranch_tcp, 5000, cowboy_clear) at pid <0.276.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.278.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.279.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.280.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.281.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.282.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.283.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.284.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.285.0>
2018-05-09 04:43:00.812 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.286.0>
2018-05-09 04:43:00.813 [debug] <0.277.0> Supervisor {<0.277.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5003>, ranch_tcp, <0.276.0>) at pid <0.287.0>
2018-05-09 04:43:00.813 [debug] <0.275.0> Supervisor {<0.275.0>,ranch_listener_sup} started ranch_acceptors_sup:start_link(http, 10, ranch_tcp, [{connection_type,supervisor},{port,8080}]) at pid <0.277.0>
2018-05-09 04:43:00.813 [debug] <0.231.0> Supervisor ranch_sup started ranch_listener_sup:start_link(http, 10, ranch_tcp, [{connection_type,supervisor},{port,8080}], cowboy_clear, #{connection_type => supervisor,env => #{dispatch => [{'_',[],[{[<<"api">>,<<"servers">>],[],...},...]}]},...}) at pid <0.275.0>
2018-05-09 04:43:00.816 [debug] <0.288.0> Supervisor {<0.288.0>,ranch_listener_sup} started ranch_conns_sup:start_link(https, supervisor, 5000, ranch_ssl, 5000, cowboy_tls) at pid <0.289.0>
2018-05-09 04:43:00.844 [debug] <0.290.0> Supervisor {<0.290.0>,ranch_acceptors_sup} started ranch_acceptor:start_link({sslsocket,nil,{#Port<0.5204>,{config,{ssl_options,tls,[{3,3},{3,2},{3,1}],verify_none,{#Fun<ssl.8...>,...},...},...}}}, ranch_ssl, <0.289.0>) at pid <0.293.0>
2018-05-09 04:43:00.845 [debug] <0.290.0> Supervisor {<0.290.0>,ranch_acceptors_sup} started ranch_acceptor:start_link({sslsocket,nil,{#Port<0.5204>,{config,{ssl_options,tls,[{3,3},{3,2},{3,1}],verify_none,{#Fun<ssl.8...>,...},...},...}}}, ranch_ssl, <0.289.0>) at pid <0.294.0>
2018-05-09 04:43:00.845 [debug] <0.290.0> Supervisor {<0.290.0>,ranch_acceptors_sup} started ranch_acceptor:start_link({sslsocket,nil,{#Port<0.5204>,{config,{ssl_options,tls,[{3,3},{3,2},{3,1}],verify_none,{#Fun<ssl.8...>,...},...},...}}}, ranch_ssl, <0.289.0>) at pid <0.295.0>
2018-05-09 04:43:00.845 [debug] <0.290.0> Supervisor {<0.290.0>,ranch_acceptors_sup} started ranch_acceptor:start_link({sslsocket,nil,{#Port<0.5204>,{config,{ssl_options,tls,[{3,3},{3,2},{3,1}],verify_none,{#Fun<ssl.8...>,...},...},...}}}, ranch_ssl, <0.289.0>) at pid <0.296.0>
2018-05-09 04:43:00.846 [info] <0.303.0> server started

gotthardp, 2018-05-09:
I believe this is related to Erlang version. You may be using an old version that does not support ciphers required by the browser. Try installing a newer Erlang.

romansoft, 2018-05-09:
When I update Erlang, do I need to recompile NS?

gotthardp, 2018-05-09:
No need to update NS. Just Erlang, perhaps also OpenSSL.

romansoft, 2018-05-09:
This does not appear to be the cause. I have the latest versions of Erlang and OpenSSL packages.
After sudo apt-get update :

[image]

I'd like to mention the network instance is running on AWS.

gotthardp, 2018-05-12:
well, in Mozilla you can disable the checks in `about:config`, by setting `network.http.spdy.enforce-tls-profile` to false.

gotthardp, 2018-05-16:
And it seems that you may need to recompile the server using Erlang 20 too.

romansoft, 2018-05-21:
I had this version already installed Erlang/OTP 20 [erts-9.3] [source] [64-bit] [smp:1:1] [ds:1:1:10] [async-threads:10] [kernel-poll:false]
I'm using precompiled NS binaries from Github, so do I need to do anything different?

I was reading that having older ciphers included in configuration can create red flags for the browsers, but I'm not sure where is the config file defining these.

Alternately, I'm exploring using nginx reverse proxy to go around this issue and make the portal more secure.

gotthardp, 2018-05-21:
That may be a build issue. You may need to rebuild the server with R20, unfortunately. Those available from GitHub are build with R19.

romansoft, 2018-05-21:
OK, this may explain it. Is ver 0.5.4 on GitHub also build with R19?

gotthardp, 2018-05-21:
Yes.

romansoft, 2018-05-22:
What I've done is set up secure Nginx reverse proxy. This works very well and is the recommended way of securing public websites. Here are the instructions (for Ubuntu and assuming previously installed Letsencrypt security certificates) in case someone is interested in this:

1. Disable SSL support in the lorawan server config file (if it was enabled before) 
2. Install nginx
3. Create a file (let's call it my_config) in /etc/nginx/sites-enabled directory/my_config
4. Run this command to generate Diffie-Hellman key (makes things more secure):
`sudo openssl dhparam -out /etc/ssl/certs/dhparam.pem 2048`
5. Open text editor and paste this content. Modify "your_domain" to match to your actual domain name
```
server {
    listen 8443 ssl;
    server_name your_domain.com;

    ssl_certificate           /etc/letsencrypt/live/your_domain.com/fullchain.pem;
    ssl_certificate_key       /etc/letsencrypt/live/your_domain.com/privkey.pem;

    ssl on;
    ssl_protocols TLSv1 TLSv1.1 TLSv1.2;
    ssl_ciphers 'EECDH+AESGCM:EDH+AESGCM:AES256+EECDH:AES256+EDH';
    ssl_prefer_server_ciphers on;
    ssl_dhparam /etc/ssl/certs/dhparam.pem;

    ssl_stapling on;
    ssl_stapling_verify on;

    add_header Strict-Transport-Security "max-age=31536000";

    access_log            /var/log/nginx/my_config.access.log;
    error_log             /var/log/nginx/error.log;

    location / {
      proxy_set_header        Host $host;
      proxy_set_header        X-Real-IP $remote_addr;
      proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
      proxy_pass          http://localhost:8080;
      proxy_read_timeout  90;
    }
}
```
6. Block your firewall on port 8080 to prevent unencrypted access to lorawan server from public internet. All interaction should be through encrypted 8443 port.
7. Restart nginx:
`sudo service nginx restart`

What it does is it handles encrypted TCP requests on port 8443 and redirects the traffic (after being decrypted) to localhost on port 8080 which is used by the lorawan server.

closed 2018-06-08 by gotthardp

hpux735, 2021-12-16:
>  6. Block your firewall on port 8080 to prevent unencrypted access to lorawan server from public internet. All interaction should be through encrypted 8443 port.
> 7. Restart nginx:
>    `sudo service nginx restart`
> 

This is a great primer, thanks for writing it.  But there's one thing I've been looking for in the documentation, and that's how to set the 8080 port to only listen on localhost.  The firewall will work, of course, but it would be better to also not listen on the external interface at all.