gotthardp/lorawan-server #389: EU868 channels 5 6 7 frequency same as 0 1 2 ?

ZuratonE, 2018-05-16:
I'm trying to add region RU870, and looking into frequency generation for EU868, found this code:
```
    if
        Ch >= 0, Ch =< 2 ->
            ch2fi(Ch, {8681, 2});
        Ch >= 3, Ch =< 7 ->
            ch2fi(Ch, {8671, 2});
        Ch == 8 ->
            868.8
    end;
```

For me, it looks like channels will generate frequencies as follows

- Ch=0: (0*2 + 8681)/10 = 868,1
- Ch=1: (1*2 + 8681)/10 = 868,3
- Ch=2: (2*2 + 8681)/10 = 868,5
- 
- Ch=3: (3*2 + 8671)/10 = 867,7
- Ch=4: (3*2 + 8671)/10 = 867,9
- Ch=5: (4*2 + 8671)/10 = 868,1 - same as Ch=0
- Ch=6: (5*2 + 8671)/10 = 868,3 - same as Ch=1 
- Ch=7: (6*2 + 8671)/10 = 868,5 - same as Ch=2
- 
- Ch=8:  868,8

Is it ok? Or should there be something like
```
        Ch >= 3, Ch =< 7 ->
            ch2fi(Ch-3, {8671, 2});
```

gotthardp, 2018-05-16:
It may be buggy, but it is not used at all. I have removed it from the code. Sorry for the confusion.
Aren't the RU frequencies identical to the EU?

ZuratonE, 2018-05-17:
Oh, thanks for clarification.

After "Regional Parameters v1.1rB" was released, frequencies differ:
RU864-870 Default channels (they're also join channels) are:
868.9MHz
869.1MHz
RX2 is 869.1MHz / DR0 (SF12, 125 kHz)

In addition to default, local operators are using channels from 864 band:
864.1MHz
864.3MHz
864.5MHz
864.7MHz
864.9MHz

gotthardp, 2018-05-17:
I see. I didn't know it was released. I will integrate the RU band in few minutes.

gotthardp, 2018-05-17:
Done. The RU868 is now added.

closed 2018-05-17 by ZuratonE