gotthardp/lorawan-server #390: parse_failed but working

webfrank, 2018-05-17:
Hi,
   today I found this error on dashboard but data is arriving without problem:

```
2018-05-17 10:14:00.816 [error] <0.1077.0> handler 5761747465636F2D43757272656E74 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:16:07.235 [error] <0.1094.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:16:10.652 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
2018-05-17 10:18:50.502 [error] <0.1111.0> handler 424154 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:20:07.246 [error] <0.1113.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:20:10.676 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
2018-05-17 10:24:07.261 [error] <0.1131.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:24:07.462 [debug] <0.1131.0>@lorawan_mac_commands:build_fopts:74 00E9229E <- [dev_status_req]
2018-05-17 10:24:10.718 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
2018-05-17 10:28:07.286 [error] <0.1160.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:28:07.486 [debug] <0.1160.0>@lorawan_mac_commands:handle_fopts:16 00E9229E -> [{dev_status_ans,255,-23}]
2018-05-17 10:28:07.486 [debug] <0.1160.0>@lorawan_mac_commands:handle_status:219 DevStatus: battery 255, margin: -23 (max -12.5)
2018-05-17 10:28:10.738 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
2018-05-17 10:32:07.279 [error] <0.1180.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:32:10.773 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
2018-05-17 10:32:55.678 [error] <0.1181.0> handler 5761747465636F2D54656D70 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:36:07.292 [error] <0.1201.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:36:10.812 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
2018-05-17 10:39:34.667 [error] <0.1228.0> handler 5761747465636F2D54656D70 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:40:07.301 [error] <0.1229.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:40:10.856 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
2018-05-17 10:44:00.775 [error] <0.1389.0> handler 5761747465636F2D43757272656E74 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:44:07.321 [error] <0.1390.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:44:10.890 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
2018-05-17 10:45:24.070 [debug] <0.298.0>@lorawan_db_guard:trim_rxframes:74 Expired rxframes from [<<"00E9229E">>,<<"0114C2A1">>,<<"0229D8CF">>,<<"0B0F0001">>]
2018-05-17 10:48:07.333 [error] <0.1408.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:48:10.911 [warning] <0.300.0> gateway 7276FF000B030A78 {ack_lost,100.0}
```
[image]

```
2018-05-17 10:14:00.816 [error] <0.1077.0> handler 5761747465636F2D43757272656E74 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:16:07.235 [error] <0.1094.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:18:50.502 [error] <0.1111.0> handler 424154 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:20:07.246 [error] <0.1113.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:24:07.261 [error] <0.1131.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:28:07.286 [error] <0.1160.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:32:07.279 [error] <0.1180.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:32:55.678 [error] <0.1181.0> handler 5761747465636F2D54656D70 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:36:07.292 [error] <0.1201.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:39:34.667 [error] <0.1228.0> handler 5761747465636F2D54656D70 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:40:07.301 [error] <0.1229.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:44:00.775 [error] <0.1389.0> handler 5761747465636F2D43757272656E74 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:44:07.321 [error] <0.1390.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:48:07.333 [error] <0.1408.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:52:07.342 [error] <0.1551.0> handler 504158 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
2018-05-17 10:53:14.287 [error] <0.1568.0> handler 424154 {parse_failed,{error,{badfun,#Fun<erl_eval.12.118419387>}}}
```

webfrank, 2018-05-17:
After re-saving all handlers I have no more parse_failed errors but I have always ack_lost

[image]

gotthardp, 2018-05-17:
This is an error in your Parse Uplink function.

webfrank, 2018-05-17:
I don't think so as data has been parsed and sent correctly and as i wrote after after saving (without modifications) every handler the error disappeared but the ack_lost not.

gotthardp, 2018-05-17:
Well, the error *is* being raised by the Parse Downlink function. The server is able to recover from that error though, so some functionality is maintained.

closed 2018-05-27 by gotthardp