gotthardp/lorawan-server #392: Gateway showing disconnected (RPi3)

gmbankar37, 2018-05-18:
Hello,
 I have my gateway on the host raspberrypi 3 and It is connected to wifi. I have completed all the necessary procedure or changes in the packet forwarder _config.json files.  But my gateway is showing status disconnected. I am not able to connect my gateway to the server. And because my gateway is not online I cant join the node. PLease  help me.

gotthardp, 2018-05-18:
Are you able to "ping" from the Raspberry to the server IP address you did put to the config.json?

gmbankar37, 2018-05-19:
Actually I am providing internet to RPi3 through mobile hotspot so the
wlan0 ip address keeps changing. Yeah I am able to ping to the wlan
address.

gmbankar37, 2018-05-19:
Do my server always have to be connected to the internet? I mean does the
packet forwarder uses TCP to forward the data from gateway to the server?
Thank you.

gotthardp, 2018-05-19:
UDP is used between the gateway and the server, so both may be disconnected for a while and the connectivity is automatically restored when they come back,

gotthardp, 2018-05-19:
If you can ping, you correctly configured both IP and also the port number of the server then it shall work.

gmbankar37, 2018-05-19:
But can we still send packet and receive it when the connectivity is not
there? Do I need to make changes in the global_conf.json file of the packet
forwarder?

gmbankar37, 2018-05-23:
> Well, of course not. If the connectivity is not there, all uplinks and
> downlinks will be lost,
> Yes, you need to set the address in the appropriate xxxx_conf.json (either
> local or global or both).

Hii,
 If I am sending a Modbus frame from node to gateway what should I prefer
internal (In Erlang) or external( handler) application ? How do I modify
the existing semtech-mote application to my requirement? Thank you.

gotthardp, 2018-05-23:
The benefits of an internal application are described at https://github.com/gotthardp/lorawan-server-demoapp

gmbankar37, 2018-06-01:
Hii, I am still facing this problem that my gateway shows disconnected. I need to solve this issue so I can do range testing. Please help.

gmbankar37, 2018-06-01:
I have solved the issue that my gateway is connected now. But I am not able to join my node. It constantly gives me 
at+recv = 6,0,0.
which is Status_RX2_TimeOUT. 
I am following the same procedure which I followed on my RPi 2. But not able to join. 
This status usually means I have configured something wrong. Are my network parameters wrong?

gotthardp, 2018-06-01:
What was the solution to the previous problem, please?

To fix the join problem, please follow the "No downlink" guide https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#no-downlink-frames-delivered

gmbankar37, 2018-06-02:
> doc/Troubleshooting.md#no-downlink-frames-delivered

I had the server and gateway setup on my RPi2 first and it had a mac id. I
used the same mac id as the previous one when I was having different host
(RPi3) to same gateway. This way it worked.

gmbankar37, 2018-06-04:
Should I change the TX chain to 1 so I will enable the radio_1. And then match my network setting to that of radio_1. So then the frequency setting in radio_1 should match with the initial RX2 freq? Am I correct? 
I am not able to understand this statement "Private networks should use 000000 or 000001" as NetID

gotthardp, 2018-06-04:
Set the TX chain to whatever radio in your gateway is configured for transmitting. Yes, the config your your radio shall match the server config.

The LoRaWAN specification requires the non-public (i.e. private networks) to use the NetID 000000 or 000001. All other NetIDs are reserved for public networks and allocated by the LoRaWAN alliance.

closed 2018-06-26 by gotthardp