gotthardp/lorawan-server #393: Downlink manual frame

jot4p, 2018-05-21:
Hi, 

It's very common in new sensors send directly one downlink frame in order to configure the sensors (like https://www.nasys.no/ sensors) 

This is necessery only one time, and I need to create a simple php callback server in order to create my downlink response/configuration.

It's possible to add new feature to server to create a simple downlink frame for sensor ? or this exists? 

I need to send a simple frame in HEX for port 50 or other .. like this (Pag 7):
https://www.nasys.no/wp-content/uploads/Pulse-Analog-Reader_UM3023_1.pdf

Thanks.

gotthardp, 2018-05-21:
Hi. The uplink that triggers this config-downlink always the same? Is it feature to "always respond to THIS uplink with THAT downlink" or to "as a response to the NEXT uplink (whatever it will be) send THIS downlink"?

jot4p, 2018-05-21:
Imagine this..  I have 1 lorawan datalogger for measure water consumption.. after first connection I need to send my settings, number of times to sync per day, number of pulses to read, read in liters  or m3, this is sended with a simple frame like: "001111001100110011001100111111111110000" (example only) or the same in HEX for a specific port (like 50 or other)
I need to send this simple command to the sensor...   maybe i need to send this at NEXT uplink

gotthardp, 2018-05-21:
How about to extend the Parse Uplink handler so it can not only forward data to a connector, but also to respond-- send immediately a downlink back? You would need to define (in Erlang) your own logic about when the downlink should be sent, or just send it always when a certain uplink is sent.

jot4p, 2018-05-22:
Hi, 
Yes, second part..  is the response for any frame only 1 time (after that sensor is configured) and I don't need to config again.

jot4p, 2018-05-23:
Ok.. I need to send this : 01A8C0830000000000000000 for the next time that sensor connects to lora server.. it's possible ?  Please give me an example. 
Thanks

gotthardp, 2018-05-23:
Sending something upon next uplink is a standard downlink feature. Yes, the image shows the frame to be sent. It was created by sending a downlink request to a connector. See https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md#downlink

jot4p, 2018-05-24:
Ok Perfect. 

It's the same send in HEX or BIN for the sensor ??

gotthardp, 2018-05-24:
you always enter binary data as a HEX string, unless the binary connector is used

hallard, 2018-05-27:
@jot4p 
I do this with sample code based on websocket example `ws.html` from the server
[basic sample](https://github.com/gotthardp/lorawan-server/blob/master/priv/admin/ws.html)
[image]

jot4p, 2018-05-28:
Thanks. Perfect.. 
I use Insomnia to create the post and it's solve my problem until next verson of your server.

jot4p, 2018-06-06:
Hi,

My downlink is processed
[image]

The downlink is sent but the sensor does not respond to my downlink.
The manufacturer confirms that it is correct (HEX):
> 00070F003C0001
>  010407028C

What is wrong ?

gotthardp, 2018-06-06:
Can you see from the device log it was received by the device?

jot4p, 2018-06-06:
No, my device is an "NAS UM3023A" 
https://www.nasys.no/wp-content/uploads/UM3023-FW-ver-0.5.0.pdf

If Isend the correct frame the device return a response.

hallard, 2018-06-06:
@jot4p I don't see any port on your downlink

[image]

No port, no data processing ;-)
Should be port 50 configuration message for your device

jot4p, 2018-06-06:
[image]

I have the port now :) (old example) 

but not work

hallard, 2018-06-06:
Looks good,
may we have the frame received screenshot when the downlink is send?

hallard, 2018-06-06:
I mean this screen for Web Interface (refreshed with communications between the server and your node)
[image]

jot4p, 2018-06-06:
[image]

hallard, 2018-06-06:
ok thanks, your device joined and sending port 99 is the starting frame of nasys devices
now the same with seeing downlink send?

jot4p, 2018-06-07:
Yes... the server send the packet and after that the sensor do not talk anymore. 

The frame "00" for port 49 make the sensor return "device frame" but I send and the sensor not return anything.

gotthardp, 2018-06-07:
If you sent a downlink (that was received by the device) and then the device stopped talking, then the server sent some ADR config data that misaligned the device configuration. Check the device profile configuration.

jot4p, 2018-06-26:
Hi, 

The creator of the sensor confirm:

´´´
Frame 1:
{
	"data": "010407020C",
	"deveui": "70B3D5B020037081",
	"port": "50"
}

Frame 2:
{
	"data": "00070F003C0001",
	"deveui": "70B3D5B020037081",
	"port": "50"
}

Confirm Frame:
{
	"data": "00",
	"deveui": "70B3D5B020037081",
	"port": "49"
} ´´´ 

[image]

After the manual procedure of sending status data (1 second with magnet), the downlink's disappear from the server (are processed and sended to the sensor ??) 

[image]

This is the response from the node correct  ? 

The creator of the sensor replied:

"It looks like the message is not received by the device. 
The second message is retransmit of the status message, as the ack is not received. If i’m correct then one f the values is fcnt and it is same for both packets. So there is no reply to fport49 message. (it should also be sent on fPort49). Is there a change you could use RX2 window instead of RX1 to send a message to the node. (in some networks it is possible to force it to RX2).  And then try again.
"

What do you think ? 

Thanks.

gotthardp, 2018-06-26:
Yes, the received frames are retransmissions. Yes, you can force RX2 in my server too (see the RX Window in the Profile setting).

jot4p, 2018-06-26:
but what is the problem ? the sensor receive the correct frame  ? How I can confirm that ?

gotthardp, 2018-06-26:
Ideally you should get debug console of the device and have a look if the downlink was received and if it was accepted by the device. Debugging devices without any logs is always hard and frustrating.

jot4p, 2018-06-26:
the default settings of the sensor:

```
ADR is supported. ADR can change the DR from SF12 to SF7. And TX power from 0-7.

The LoRaWAN setting are according to the LoRaWAN regional specs.

RECEIVE_DELAY1 1 s 
RECEIVE_DELAY2 2 s (must be RECEIVE_DELAY1 + 1s) 
JOIN_ACCEPT_DELAY1 5 s 
JOIN_ACCEPT_DELAY2 6 s 
MAX_FCNT_GAP 16384 
ADR_ACK_LIMIT 64 
ADR_ACK_DELAY 32 
ACK_TIMEOUT 2 +/- 1 s (random delay between 1 and 3 seconds) 

```

gotthardp, 2018-06-26:
What server version are you using?
As a safety measure I'd recommend to set "Set Rx2 freq" field to the default value of your Network. There is a bug (I didn't find yet) that may cause some issues.

jot4p, 2018-06-27:
My version is 0.53 
I dont set Rx2 value this is automatic ...  you can see here:
https://user-images.githubusercontent.com/15892047/41917685-e0703622-7952-11e8-869f-8e91e3709965.png

gotthardp, 2018-06-27:
I recommend you set this value too.

jot4p, 2018-07-02:
I remove my device and create new entry on commisioned list:

[image]

After that (configured with manufacture settings) my device try to join but is not possible !! 
why ?  
[image]

[image]

jot4p, 2018-07-02:
my config in multitech packet forwarder
```

{
	"SX1301_conf": {
		"antenna_gain": 8,
		"chan_FSK": {
			"bandwidth": 125000,
			"datarate": 50000,
			"enable": true,
			"if": 300000,
			"radio": 1
		},
		"chan_Lora_std": {
			"bandwidth": 250000,
			"enable": true,
			"if": -200000,
			"radio": 1,
			"spread_factor": 7
		},
		"chan_multiSF_0": {
			"enable": true,
			"if": -400000,
			"radio": 1
		},
		"chan_multiSF_1": {
			"enable": true,
			"if": -200000,
			"radio": 1
		},
		"chan_multiSF_2": {
			"enable": true,
			"if": 0,
			"radio": 1
		},
		"chan_multiSF_3": {
			"enable": true,
			"if": -400000,
			"radio": 0
		},
		"chan_multiSF_4": {
			"enable": true,
			"if": -200000,
			"radio": 0
		},
		"chan_multiSF_5": {
			"enable": true,
			"if": 0,
			"radio": 0
		},
		"chan_multiSF_6": {
			"enable": true,
			"if": 200000,
			"radio": 0
		},
		"chan_multiSF_7": {
			"enable": true,
			"if": 400000,
			"radio": 0
		},
		"clksrc": 0,
		"lbt_cfg": {
			"enable": false,
			"rssi_target": 160
		},
		"lorawan_public": true,
		"radio_0": {
			"enable": true,
			"freq": 867500000,
			"rssi_offset": -162,
			"tx_enable": true,
			"tx_freq_max": 870000000,
			"tx_freq_min": 863000000,
			"type": "SX1257"
		},
		"radio_1": {
			"enable": true,
			"freq": 868500000,
			"rssi_offset": -162,
			"tx_enable": true,
			"type": "SX1257"
		},
		"tx_lut_0": {
			"dig_gain": 0,
			"mix_gain": 11,
			"pa_gain": 0,
			"rf_power": -6
		},
		"tx_lut_1": {
			"dig_gain": 0,
			"mix_gain": 13,
			"pa_gain": 0,
			"rf_power": -3
		},
		"tx_lut_10": {
			"dig_gain": 0,
			"mix_gain": 15,
			"pa_gain": 2,
			"rf_power": 16
		},
		"tx_lut_11": {
			"dig_gain": 0,
			"mix_gain": 10,
			"pa_gain": 3,
			"rf_power": 20
		},
		"tx_lut_12": {
			"dig_gain": 0,
			"mix_gain": 12,
			"pa_gain": 3,
			"rf_power": 23
		},
		"tx_lut_13": {
			"dig_gain": 0,
			"mix_gain": 13,
			"pa_gain": 3,
			"rf_power": 25
		},
		"tx_lut_14": {
			"dig_gain": 0,
			"mix_gain": 15,
			"pa_gain": 3,
			"rf_power": 26
		},
		"tx_lut_15": {
			"dig_gain": 0,
			"mix_gain": 15,
			"pa_gain": 3,
			"rf_power": 27
		},
		"tx_lut_2": {
			"dig_gain": 0,
			"mix_gain": 9,
			"pa_gain": 1,
			"rf_power": 0
		},
		"tx_lut_3": {
			"dig_gain": 0,
			"mix_gain": 10,
			"pa_gain": 1,
			"rf_power": 3
		},
		"tx_lut_4": {
			"dig_gain": 0,
			"mix_gain": 12,
			"pa_gain": 1,
			"rf_power": 6
		},
		"tx_lut_5": {
			"dig_gain": 0,
			"mix_gain": 10,
			"pa_gain": 2,
			"rf_power": 10
		},
		"tx_lut_6": {
			"dig_gain": 0,
			"mix_gain": 11,
			"pa_gain": 2,
			"rf_power": 11
		},
		"tx_lut_7": {
			"dig_gain": 0,
			"mix_gain": 11,
			"pa_gain": 2,
			"rf_power": 12
		},
		"tx_lut_8": {
			"dig_gain": 2,
			"mix_gain": 12,
			"pa_gain": 2,
			"rf_power": 13
		},
		"tx_lut_9": {
			"dig_gain": 0,
			"mix_gain": 13,
			"pa_gain": 2,
			"rf_power": 14
		}
	},
	"gateway_conf": {
		"forward_crc_disabled": false,
		"forward_crc_error": false,
		"forward_crc_valid": true,
		"gateway_ID": "00800000A00005C9",
		"keepalive_interval": 10,
		"push_timeout_ms": 100,
		"serv_port_down": 1680,
		"serv_port_up": 1680,
		"server_address": "loraserver.dstinovacao.com",
		"stat_interval": 30,
                 "autoquit_threshold": 60
	}
}
```