gotthardp/lorawan-server #397: Cannot connect to connector if 2 connectors is define and one handler.

Gligor721, 2018-05-23:
I have one Handler and 2 connectors.
My connectors:
  | WEBSOCKET_CONNECTOR | WEBSOCKET | ws: | /ws/uplink/{app} 
  | SENDMESSAGE | WEBSOCKET | ws: | /ws/uplink/{devaddr}  
My handler:
  | WEBSOCKET | deveui devaddr port fcnt data | ascii | never
When my 2 connectors is enabled I can open websocket with ws://10.10.5.246:8080/ws/uplink/00200047 but when i cannot open with ws://10.10.5.246:8080/ws/uplink/WEBSOCKET.
If I DISABLE the sendmessage connector then I can open the websocket and then if I ENABLE send message connector I can open the websocket.
If two connectors is enabled I can open only sendmessage connector but not WEBSOCKET_CONNECTOR.If i disable sendmesage I can open WEBSOCKET_CONNECTOR and is working and then I need to ENABLE SENDMESSAGE.
Why is this happening?

Gligor721, 2018-05-24:
Ok I understand.Now how to solve this problem.I need one connector so my app get app packets from all node and if I want to send packet to node(one node)I need one connector so I can send packet to the node.if I send packet to connector that received all node, packet will be send to all nodes I dont whant that and that is the reason for my another connector.
Please tell me how to solve my problem?

gotthardp, 2018-05-24:
Well, simply use two different paths, e.g.
  | WEBSOCKET_CONNECTOR | WEBSOCKET | ws: | /ws/uplink1/{app}
  | SENDMESSAGE | WEBSOCKET | ws: | /ws/uplink2/{devaddr}

Gligor721, 2018-05-24:
Ok I think that I cannot change uplink name in path.

Gligor721, 2018-05-24:
You want to say that I can use /ws/uplink/what_I_want/{one of define in documentation} so in that way I can define multiple connectors?for example one connector /ws/uplink/allnode/{app} second connector /ws/uplink/onenode/{devaddr} etc.What about events?

gotthardp, 2018-05-24:
It's even /what/you/want. It has to start with a slash and the static part has to be unique for the entire server. It doesn't have to start with /ws, it can have more defines and the last define can be followed by a static suffix. The same for events: start with a slash and be unique.

closed 2018-06-08 by gotthardp