gotthardp/lorawan-server #398: localhost refused connection

gmbankar37, 2018-05-24:
hii, 
 I was able to run the lorawan server and was manipulating the nodes through Web administration on localhost:8080 . But suddenly when I started the server today it is showing me that page is not found and localhost refused the connection. Yesterday I changed the sys.conf fille to add new application. Is it because of that? 
please help me. thank you.

gotthardp, 2018-05-24:
And does it when you revert the change? It's likely a misconfiguration.

gmbankar37, 2018-05-24:
Yes I figured it out and reverted back to the original. But I should be
able to have more than one internal application right? Also where is the
received frames stored in the host system (I am using raspberrypi). Thank
you.

gotthardp, 2018-05-24:
The received frames are stored in a database. The database is in /var/lib/lorawan-server.

And yes, you shall be able to have multiple internal apps.

gmbankar37, 2018-05-28:
Hi,
 I am a little confused which database is used to store RX frame ? Mnesia
(Erlang) or MongoDB (as said in readme file). Also how can I send
(/extract) my data to an application server through API ? Thank you.

gotthardp, 2018-05-28:
Did you solve the original issue?

All RX frames are stored in the Mnesia database. This you can view via web-admin and access via REST API `/api/rxframes`. See https://github.com/gotthardp/lorawan-server/blob/master/doc/Administration.md#rest-api

You can also create a MongoDB connector that will store frames from some (or all your) devices into a MongoDB. The server does not provide any access to these data; you have to use a standard MongoDB functions or develop your own.

gmbankar37, 2018-05-28:
> doc/Administration.md#rest-api

Yes I have solved the previous issue. okay I just want to store the data on
an application server from where I can retrieve it and manipulate it. Thank
you.

closed 2018-05-28 by gotthardp