gotthardp/lorawan-server #399: How to setup EU868 all 8 Channels

hallard, 2018-05-24:
I needed to use all EU868 channels but I don't know how to handle them

My lopy is set as follow and with OTAA
[image]

Do I need to do anything on server (0.5.4) and if so, how do I add them?

Is it something like that, not sure on value format to put in Min/Max Data Rate
[image]

Thanks

gotthardp, 2018-05-24:
Yes, this is exactly the setting you need to provide. It's even identical to your lopy config! Do I really need to explain that your "dr_min" requals to my "Min Data Rate"?
Note also that you have to set "Initial Channels" to `0-7`.

gotthardp, 2018-05-24:
Note also that your frequency is in Hz, but my is in MHz. Make sure you use correct units.

hallard, 2018-05-24:
Correct, got it thanks,
As documentation for data rate state " If not specified, equals to the global value on the ADR tab." and on the ADR tab it's like that 
[image]
I was not sure what format to use.

closed 2018-05-24 by hallard