gotthardp/lorawan-server #401: Please help me with setuping lorawan network through windows

shyam-coder, 2018-05-26:
Before submitting a bug report please
 - read the Troubleshooting documentation;
 - make sure you are running the latest stable release;
 - inspect the debug.log to check if there are some warnings or errors reported.

shyam-coder, 2018-05-26:
I am unable to find the way to set

gotthardp, 2018-05-26:
If you need more detailed answer, please provide more detailed question.

shyam-coder, 2018-05-26:
Thank you I will try

shyam-coder, 2018-05-26:
thank you so much bro

shyam-coder, 2018-05-26:
my question is Iam currently having a gateway and nodes based on arduino
,so when I see in loraserver.io website it have instructions and links for
ubuntu so I was need of help to do the same setup with windows.

shyam-coder, 2018-05-26:
gateway based on pic

gotthardp, 2018-05-26:
loraserver.io is a different server
the instructions on how to install my server on windows are here: https://github.com/gotthardp/lorawan-server/blob/master/doc/Installation.md#using-the-binary-release-on-windows

shyam-coder, 2018-05-26:
> doc/Installation.md#using-the-binary-release-on-windows

Oh sorry, thanks again.

shyam-coder, 2018-05-27:
how to install gateway,appserver in windows,server worked well

gotthardp, 2018-05-27:
I cannot help you with that. Installation of the gateway software depends on the hardware you have. Please contact your gateway supplier.

shyam-coder, 2018-05-27:
Ok

closed 2018-05-27 by gotthardp