gotthardp/lorawan-server #402: Server crashed !!

gmbankar37, 2018-05-29:
Before submitting a bug report please
 - read the Troubleshooting documentation;
 - make sure you are running the latest stable release;
 - inspect the debug.log to check if there are some warnings or errors reported.

gmbankar37, 2018-05-29:
Hello, 
  My server crashed and I am not able to load/start it again. The crash report is as
[crash.log](https://github.com/gotthardp/lorawan-server/files/2046869/crash.log) . Do I need to build the server again?

gotthardp, 2018-05-29:
You made an error in the configuration. Your "MQTT_PI" has a pattern "out/{00000AB1}", but "00000AB1" is not a valid pattern name. Hence the crash. It shall not be so devastating (I will fix that), but the rootcause is the misconfiguration. You can try stopping the server, deleting the "connectors" database file and restarting the server again.

gmbankar37, 2018-05-29:
Yes I will do that. So you want me to delete connectors.DCD file. But I followed the document and “out/{devaddr}” is the pattern given and 00000AB1 is my devaddr. Is my devaddr itself incorrect? 
 

From: Petr Gotthard
Sent: Tuesday, May 29, 2018 12:32 PM
To: gotthardp/lorawan-server
Cc: gmbankar37; Author
Subject: Re: [gotthardp/lorawan-server] Server crashed !! (#402)

You made an error in the configuration. Your "MQTT_PI" has a pattern "out/{00000AB1}", but "00000AB1" is not a valid pattern name. Hence the crash. It shall not be so devastating (I will fix that), but the rootcause is the misconfiguration. You can try stopping the server, deleting the "connectors" database file and restarting the server again.

gotthardp, 2018-05-29:
Yes, please-- delete the connectors.DCD file
It's a pattern, so you shall use `{devaddr}` as is. The devaddr will be replaced automatically by the server.

gotthardp, 2018-06-05:
Will be more robust in v0.5.5

closed 2018-06-05 by gotthardp