gotthardp/lorawan-server #403: Debug log file not updating

gmbankar37, 2018-05-30:
hii,
 My debug log file is not updating though crash and error files are working fine. Here is the code 
in my sys.config file. 
% error logging settings
{lager, [
    {colored, true},
    % uncomment to display even a high number of messages per second
    % see https://github.com/erlang-lager/lager#overload-protection
    % {error_logger_hwm, undefined},
    {handlers, [
        {lager_console_backend, [{level, debug}]},
        {lager_file_backend, [{file, "debug.log"}, {level, debug}]},
        {lager_file_backend, [{file, "error.log"}, {level, error}]}
    ]},
    {crash_log, "crash.log"}
]},
{sasl, [
    {errlog_type, error}
]}].
please help me. Thank you.

gotthardp, 2018-05-30:
It seems that the process does not have access to that file. Wrong permissions? Disk full?

gmbankar37, 2018-05-30:
It shows the disk_almost_full. How can I solve this problem?

closed 2018-06-08 by gotthardp